mavlink/qgroundcontrol #11354: Issue with QGC guide (docs) webpage

omid-esrafilian, 2024-04-19:
@DonLakeFlyer 
It appears that there is an issue with the QGC guide (docs) webpage not loading correctly. 
Please check the link below:
https://docs.qgroundcontrol.com/master/en/qgc-user-guide/index.html

I have also provided a screenshot below:

[image]

HTRamsey, 2024-04-19:
I think the docs workflow needs to be re-run after reverting my changes in #11348 which requires a new push. Or maybe we can change it such that manual runs cause the deploy job to be run too.

HTRamsey, 2024-04-20:
Nevermind, it's not related to workflow. Maybe something with the docs changes in your PR I guess.

omid-esrafilian, 2024-04-20:
@HTRamsey 
Thanks for the response. But it seems that the problem still persists.
I am not an expert in web development, but as an idea, shall we create a pull request with some minor changes in docs to find out what seems to be the problem, and if there are any compile errors pertaining to docs?

HTRamsey, 2024-04-20:
I just built it locally on the latest master and it works fine, maybe something server side no idea

closed 2024-04-20 by HTRamsey

omid-esrafilian, 2024-04-20:
@HTRamsey 
Great, thanks much.