mavlink/qgroundcontrol #11408: I am unable to find qgroundcontrol.pro file for custom development

zerocool-z1, 2024-04-29:
i have followed all the steps mentioned in qgc development guide. i downloaded stable 4.4 brnd it had qgroundcontrol.pro file. but it required 5.15 qt version which is not available now. the thing is right now i am unable to find qgc.pro file in the master branch, how should i proceed for the custom development. a month back everything was fine and i was able to download repository the just buy the command git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git 
and it contained all the files but now it's missing please help....

HTRamsey, 2024-04-29:
I have restored those files. Just FYI, qmake is being replaced by a cmake build system and isn't guaranteed to provide the same functionality.

closed 2024-04-29 by HTRamsey

DonLakeFlyer, 2024-04-29:
Also FYI: Since we are in the middle of a build system transition the current custom_example custom build stuff isn't currently working with qmake. If you work at it you would be able to fix it up and make it work though. It's just not working without changes right now. We'll fix that up soon.

zerocool-z1, 2024-04-30:
Thank you so much

zerocool-z1, 2024-04-30:
also can you please provide me any guide to make android qgc on macbook

DonLakeFlyer, 2024-04-30:
> also can you please provide me any guide to make android qgc on macbook

Use the Qt docs