mavlink/qgroundcontrol #11414: member access into perameter type

zerocool-z1, 2024-04-30:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

zerocool-z1, 2024-04-30:
/Users/anshadlakha/qgroundcontrol/custom/src/AutoPilotPlugin/CustomAutoPilotPlugin.cc:38: error: member access into incomplete type 'ParameterManager'
/Users/anshadlakha/qgroundcontrol/custom/src/AutoPilotPlugin/CustomAutoPilotPlugin.cc:38:45: error: member access into incomplete type 'ParameterManager'
            if (_vehicle->parameterManager()->parametersReady()) {
                                            ^
/Users/anshadlakha/qgroundcontrol/src/Vehicle/Actuators/Common.h:22:7: note: forward declaration of 'ParameterManager'
class ParameterManager;
      ^

HTRamsey, 2024-04-30:
just include ParameterManager.h

zerocool-z1, 2024-04-30:
it is already included

HTRamsey, 2024-04-30:
in CustomAutoPilotPlugin.cc?

closed 2024-05-07 by HTRamsey