mavlink/qgroundcontrol #11437: Update Docs for CMake

HTRamsey, 2024-05-01:


DonLakeFlyer, 2024-05-05:
> * The warning on the Licensing page is incorrect

What part? It just seems to be a repeat of what is said above it: "_QGroundControl_ (QGC) is dual-licensed as Apache 2.0 and GPLv3. All contributions have to be made under both licenses. Users of the codebase are free to use it under either license."

I don't think it's really needed, but not sure what part you think is wrong?

HTRamsey, 2024-05-05:
The part where it says "QGroundControl licensing rules out the re-use of any copyleft (e.g. GPL) licensed code. All contributions must be original or from a compatible license (BSD 2/3 clause, MIT, Apache 2.0).".
It doesn't make sense to 'rule out' using GPL when it is GPL licensed itself.

smadi0x86, 2024-06-22:
I would like to also note that the command in the QGroundControl documentation for building the project with CMake cli has a missing flag:

**This command should include the -D flag:**

```c
cmake -B build -G Ninja -DCMAKE_BUILD_TYPE=Debug
```
#### Proposed Solution:

Update the documentation to use the correct CMake command with the -D flag.

Without the `-D` flag, CMake does not interpret `CMAKE_BUILD_TYPE=Debug` as a variable definition. Instead, it treats it as an additional path or parameter, leading to this error:

```
CMake Error: The source directory "~/dev/Personal/qgroundcontrol/CMAKE_BUILD_TYPE=Debug" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
```

**Additional Information:**

This worked for me to build using cmake cli, however, I might be doing something wrong, so please let me know if there are any additional steps or corrections needed.

DonLakeFlyer, 2024-08-20:
> * I would remove the centos 7 page. There is nothing particularly unique about the centos process, also centos 7 is EOL.

Done

> * I dont think there are dev calls so maybe remove that?

Done

> This command should include the -D flag:

Done