mavlink/qgroundcontrol #11457: Broken build

TishSerg, 2024-05-03:
## Expected Behavior
Successful build

## Current Behavior
Build failed

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. `git checkout f4b977b`
2. Try rebuild
3. See successful build
4. `git checkout 623b145`
5. Try rebuild
6. See failed build with errors "jom: ... Makefile [UnitTestList.obj] Error 2" and "jom: ... Makefile [InitialConnectTest.obj] Error 2"
7. `git checkout 2352de3`
8. Try rebuild
9. See failed build with errors "src\FirmwarePlugin\FirmwarePlugin.cc(12): fatal error C1083: Cannot open include file: 'GenericAutoPilotPlugin.h': No such file or directory" and "jom: ... Makefile [FirmwarePlugin.obj] Error 2"

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11 23H2
- QGC Version: master
- QGC build: daily, self-built from source

## Detailed Description
The latest two commits to master (2352de3 and 623b145) have broken build for me. Latest working commit is f4b977b.
@HTRamsey it's your today's latest commits. Could you take a look?

HTRamsey, 2024-05-03:
Is that with qmake? Everything is swapped over to cmake so it gets hard to keep them both working consistently. I will fix it real quick.

TishSerg, 2024-05-03:
> Is that with qmake?

Yes
[image]

HTRamsey, 2024-05-03:
Should be good

TishSerg, 2024-05-03:
> Should be good

Yes, I just successfully built on 2af81a6.
Thank you!

closed 2024-05-03 by TishSerg