mavlink/qgroundcontrol #11474: Conflict between QGC's "Geo" and the origional `GeographicLib` library

changx03, 2024-05-05:
I need to use the utility functions in the `GeographicLib` library. I imported it as a static library on Windows.
However, the code from `src/Geo` reimplements part of the GeographicLib library with many changes, e.g, replacing `Math::real` as `double`, etc.
As a result, when I build it from Qt Creator, the complier reports errors like these:

```
LNK2005: "public: static int __cdecl GeographicLib::Math::digits10(void)" (?digits10@Math@GeographicLib@@SAHXZ) already defined in GeographicLib.lib(Math.obj)
```

There are many of similar errors in `Math.hpp`. All files under `src/Geo` use the namespace as `GeographicLib` but don't implement all the functions.

The class I want to use is `GeographicLib::Geodesic`.  Is it a way to get around this?

Thanks.

DonLakeFlyer, 2024-05-05:
Are you using the latest daily build and cmake? You should be able to tell cmake what version of the Geo library you want if you want a newer one that QGC is currently using. You shouldn't need to import Geo again as a separate library.

HTRamsey, 2024-05-05:
Yeah, the update I just added in #[11466](https://github.com/mavlink/qgroundcontrol/pull/11466) should let you use whatever version you want now, or use a previously installed version.

changx03, 2024-05-06:
I forked Stable 4.2. I noticed the conflict exits even in the Stable 4.4 branch. Is this a future release? Thanks.

HTRamsey, 2024-05-06:
Yeah it's in the current development branch that will end up being v5.0

HTRamsey, 2024-05-07:
I'm going to close this since it is already handled in the development branch.

closed 2024-05-07 by HTRamsey