mavlink/qgroundcontrol #11501: Apparent permission issues on generated binary file (no maps loaded either)

TSC21, 2024-05-07:
## Expected Behavior
No permission issues, network access and maps loading.

## Current Behavior
I do not get any maps loaded and apparently there's no network access.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Clone current `master` (as per May 7, 2024)
2. (Debug) Build it using Qt Creator (I am building it in Ubuntu 22.04)
3. Try to run it directly from Qt Creator or using the CLI (with `./QGroundControl --verbose`)

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04
- QGC Version: current `master`
- QGC build: self-built from source
- Flight Controller: N.A.
- Autopilot (with version): N.A.

## Detailed Description
I tried to change some user permissions to things like GeoClue2, but without luck. I am assuming this is a permissions issue in the generated binary. I am QGroundControl [v4.3.0](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.3.0) AppImage without any issues. I am building using v6.6.3 build kit.

## Log Files and Screenshots

QGC console log:
```console
./QGroundControl --verbose
QML debugging is enabled. Only use this in a safe environment.
Settings location "/home/nuno/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nqgc.*.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
qt.multimedia.ffmpeg.libsymbolsresolver: Couldn't load VAAPI library
MAVLinkLogManagerLog: MAVLink logs directory: "/home/nuno/Documents/QGroundControl Daily/Logs"
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
Map Cache in: "/home/nuno/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error QGeoPositionInfoSource::AccessError
QGeoTileRequestManager: Failed to fetch tile (1,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,0,2) 5 times, giving up. Last error message was: 'Network not available'
```

TSC21, 2024-05-07:
I don't know necessarily if this is a bug or something missing on my setup or configuration, so I would appreciate some advice here. Thanks!

HTRamsey, 2024-05-07:
I googled "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000" and found a lot of results with similar solutions. Try those and let me know. I tried changing the internet detection method to a more 'direct' method but apparently it requires more permissions for some people.

TSC21, 2024-05-07:
> I googled "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000" and found a lot of results with similar solutions. Try those and let me know. I tried changing the internet detection method to a more 'direct' method but apparently it requires more permissions for some people.
> 
> I may have to add a fallback internet check method when permissions checks fail.

Yes the geolocation part is easy to figure out by just enabling it in the GNOME Privacy settings. But the network access is still a problem.

HTRamsey, 2024-05-07:
Do you know if you are using systemd-networkd or NetworkManager

TSC21, 2024-05-07:
> Do you know if you are using systemd-networkd or NetworkManager

NetworkManager. One thing I can say is that it seems that `QNetworkInformation::instance()` singleton is never created.

HTRamsey, 2024-05-07:
Hmm okay odd. I've not seen that issue come up before but I will add a simple backup for when QNetworkInformation doesn't work.

DonLakeFlyer, 2024-05-07:
Is it with all map providers or just a specific one?

TSC21, 2024-05-07:
> Hmm okay odd. I've not seen that issue come up before but I will add a simple backup for when QNetworkInformation doesn't work.

Is it reall the solution to implement workarounds to the Qt plugin back-end functionalities instead of maybe trying to understand why they fail in the first place? Can it be that the plugin is just not being loaded?

TSC21, 2024-05-07:
> Is it with all map providers or just a specific one?

It seems map unrelated (I tried selecting other map sources and saw the same issue). I do believe this is related to a a backend plugin either failing to load or some permission issue.

TSC21, 2024-05-07:
When using `QT_DEBUG_PLUGINS=1`, it appears that the plugin is indeed loading:

```console
qt.core.plugin.factoryloader: checking directory path "/home/nuno/Firestorm/qgroundcontrol/build/Desktop_Qt_6_6_3_GCC_64bit-Debug/staging/Qt/plugins/networkinformation" ...
qt.core.plugin.factoryloader: checking directory path "/home/nuno/Firestorm/qgroundcontrol/build/Desktop_Qt_6_6_3_GCC_64bit-Debug/staging/networkinformation" ...
```

But then it fails to load the default backend. Tried to load by name using `networkmanager`, with the same result.

HTRamsey, 2024-05-07:
If you are still building with qmake instead of cmake then that would makes sense as to why it isn't loading the plugin

DonLakeFlyer, 2024-05-07:
You would be getting a plugin load failure message is the plugin wasn't loading. Doesn't seem like you are getting that. Can you download firmwares for flashing?

TSC21, 2024-05-07:
> ```
> networkinformation.so
> ```

Did we completely deprecate the usage of qmake? I mean AFAIK that has been the default since I started developing for QGC and the documentation doesn't tell anywhere yet to replace `qmake` build commands with `cmake` build commands in Qt Creator.

> You would normally be getting a plugin load failure message if the plugin wasn't loading. Doesn't seem like you are getting that.

I was actually expecting something as it is showing in this comment: https://forum.qt.io/topic/144062/qt6-4-3-qnetworkinformation-not-working/7?_=1715114008429&lang=en-US. Seems like any plugin found does provide metadata. I actually don't see this plugin anywhere on my system. Am I missing something?

> Can you download firmwares for flashing? Are you behind a proxy?

I am not behind a proxy and I don't have a flight controller at hand to try that. But what I can say is that I don't see this issue in v4.3.

TSC21, 2024-05-07:
Fixed in https://github.com/mavlink/qgroundcontrol/pull/11502. Thanks!

closed 2024-05-07 by TSC21

DonLakeFlyer, 2024-05-08:
> I mean AFAIK that has been the default since I started developing for QGC and the documentation doesn't tell anywhere yet to replace `qmake` build commands with `cmake` build commands in Qt Creator.

Look at the latest docs. Switching to cmake instead of qmake has been in the works forever. The dev guide was updated a few days back with cmake info.

DonLakeFlyer, 2024-05-08:
> Did we completely deprecate the usage of qmake?

For regular builds, that is essentially true. For custom builds, we will still be on qmake for a little longer till we finish that up.

TSC21, 2024-05-08:
@DonLakeFlyer sounds good. Thanks for clarifying!

DonLakeFlyer, 2024-05-08:
The docs have been a bit behind what has been going on in the source code.