mavlink/qgroundcontrol #11506: Why QGC load things from D drive?

lida2003, 2024-05-09:
As the qgc software is installed in C drive. And I have never use anyconfiguration on drive D. 

So why it has log related wit D?

And it's really slow on windows, is it something to do with the code itself?
- https://github.com/mavlink/qgroundcontrol/issues/10126

```
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:285 - "Settings location "C:/Users/Administrator/AppData/Roaming/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true"
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCLoggingCategory.cc:122 - "Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false""
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:388 - "System reported locale: QLocale(Chinese, Simplified Han, China) ; Name "zh_CN" ; Preffered (used in maps):  "en-US""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:413 - "Qt lib localization for "zh_CN" is not present"
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:418 - "Error loading source localization for "zh_CN""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:423 - "Error loading json localization for "zh_CN""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\PositionManager\PositionManager.cpp:201 - "QGCPositionManager error 0"
[E] at :0 - "Location access failed."
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\Vehicle\MAVLinkLogManager.cc:360 - "MAVLink logs directory: "C:/Users/Administrator/Documents/QGroundControl Daily/Logs""
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGCMapEngine.cpp:164 - "Map Cache in: "C:/Users/Administrator/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[D] at qrc:/qml/FlightDisplayViewUVC.qml:54 - "UVC Video output visible:  false"
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\MissionManager\MissionController.cc:2371 - "setCurrentPlanViewSeqNum"
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\MissionManager\MissionController.cc:2371 - "setCurrentPlanViewSeqNum"
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t3.tiles.virtualearth.net/tiles/a132121030313202303.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t2.tiles.virtualearth.net/tiles/a132121030313203230.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313202313.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313221001.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t3.tiles.virtualearth.net/tiles/a132121030313221011.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313203320.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313202332.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313203221.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t2.tiles.virtualearth.net/tiles/a132121030313202.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t3.tiles.virtualearth.net/tiles/a132121030313203211.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313203213.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313202300.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313203200.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313202333.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313220101.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313202301.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313202312.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t3.tiles.virtualearth.net/tiles/a132121030313203223.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313203212.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t2.tiles.virtualearth.net/tiles/a132121030313203202.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313203213.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313203221.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313203320.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313220101.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t0.tiles.virtualearth.net/tiles/a132121030313202312.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QtLocationPlugin\QGeoMapReplyQGC.cpp:180 - "Fetch tile error: "Error transferring http://ecn.t1.tiles.virtualearth.net/tiles/a132121030313220101.jpeg?g=563&mkt=en-US - server replied: Bad Gateway""
[... cut ...]
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:388 - "System reported locale: QLocale(Chinese, Simplified Han, China) ; Name "zh_CN" ; Preffered (used in maps):  "en-US""
[E] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:413 - "Qt lib localization for "az_AZ" is not present"
[D] at D:\a\qgroundcontrol\qgroundcontrol\src\QGCApplication.cc:388 - "System reported locale: QLocale(Chinese, Simplified Han, China) ; Name "zh_CN" ; Preffered (used in maps):  "en-US""
[E] at qrc:/qml/OfflineMap.qml:218 - "qrc:/qml/OfflineMap.qml:218:9: Cannot assign to non-existent property "defaultSuffix""

```

HTRamsey, 2024-05-09:
Caching location is set automatically by Qt using [this](https://doc.qt.io/qt-6/qstandardpaths.html#StandardLocation-enum). And Windows 11 isn't supported officially until Qt 6.7. We are on 6.6.

lida2003, 2024-05-09:
> Caching location is set automatically by Qt using [this](https://doc.qt.io/qt-6/qstandardpaths.html#StandardLocation-enum). And Windows 11 isn't supported officially until Qt 6.7. We are on 6.6.

There is no drive D in the [link](https://doc.qt.io/qt-6/qstandardpaths.html#StandardLocation-enum). What I have missed here?

And any way to solve this issue? install QT6.6? Well, I'm NOT expert on this. So those APIs doesn't help a lot. Is there any way to fix the issue?

closed 2024-10-16 by HTRamsey