mavlink/qgroundcontrol #11513: I am having trouble getting my Radiomaster TX12 MKII CC2500 MPM transmitter and RadioMaster R88 V2 8CH receiver connected with the Pixhawk 2.4.8. Does have one have any suggestions on the physical connection that is best and the settigs I should use? did research int he Wiki

timemaker1, 2024-05-10:
Tell us a bit about the feature:
- What problem does it solve?
- What does it do?
- What flight stacks must it work with? (All, PX4, ArduPilot)
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
- Are there other systems that have this feature?
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)

booo, 2024-06-12:
Unrelated to QGroundControl

closed 2024-06-12 by booo

wikstorm, 2024-09-06:
@booo Did you manage to solve this? I cannot get my autopilot running PX4 to find the Radiomaster R88.

timemaker1, 2024-09-08:
hello there. no I did not solve it. I tried and still have not had any
success after purchasing two different controllers.