mavlink/qgroundcontrol #11516: Build is broken again

TishSerg, 2024-05-10:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Successful build

## Current Behavior
Build failed

## Steps to Reproduce:
1. `git checkout 8f6f4c6`
2. Try rebuild
3. See failed build

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11 23H2
- QGC Version: master
- QGC build: daily, self-built from source

## Detailed Description
The 8f6f4c6 commit to master have broken build for me. Latest working commit is 35f8864.
@HTRamsey that yours.

HTRamsey, 2024-05-10:
We don't have a CI build that checks qmake anymore. I would recommend using cmake to avoid this

closed 2024-05-10 by HTRamsey

TishSerg, 2024-05-10:
> I would recommend using cmake to avoid this.

I set up build environment according to official instructions. That use qmake.
How to make transition to cmake?