mavlink/qgroundcontrol #11518: Video settings dissapear

gendebeat, 2024-05-10:
Hello, after a while of starting the application (15 seconds), the video input stops and if I go to application settings, I can't see the video settings, it disappears!.

Another thing that has been happening to me for several versions is that in the application menu i see gliches in the interface.
[image]
Thanks

gendebeat, 2024-05-10:
i have thjis message in console
[image]

HTRamsey, 2024-05-10:
Is this on v4.3?

gendebeat, 2024-05-13:
yes. and in the previous. I think there may be some conflict on my
computer, with gstreamer (which uninstalled it) or the gpu.
but I am not able to know why

closed 2025-05-03 by HTRamsey