mavlink/qgroundcontrol #11523: QGroundControl Crashes with H.265 Stream When AutoConnect is Enabled

nechtobolshee, 2024-05-13:
## Expected Behavior
An H.265 encoded video stream should be successfully transmitted and displayed in QGroundControl even when AutoConnect is enabled.

## Current Behavior
When AutoConnect is enabled in QGroundControl, the application crashes when attempting to accept and display an H.265 encoded video stream. H.264 video stream with AutoConnect enabled works fine in all cases.

## Steps to Reproduce:
1. Prepare a video stream in H.265 format.
2. Configure QGroundControl to accept an H.265 video stream and disable all AutoConnect settings.
3. Start streaming video
4. Check that the video stream is successfully displayed in QGroundControl.
5. Stop video streaming
6. Enable the AutoConnect setting in QGroundControl.
7. Try to transmit the video stream in H.265 format. The application crashes.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04
- QGC Version: master
- QGC build: daily
- Flight Controller and Autopilot are not important in this case.

## Detailed Description
In our internal pipeline, a video stream is processed and then encoded in H.265 format before being sent to QGroundControl for display. When all automatic connections (AutoConnect) in QGroundControl are disabled, and the video stream settings are set to use the H.265 codec, the video displays successfully without any issues. However, when the AutoConnect feature is enabled, QGroundControl struggles with the stream in this format, and the program crashes.

On the other hand, when using the H.264 codec, regardless of the AutoConnect feature's state, the video stream is processed and displayed correctly without any failures. This suggests that the problem lies in the compatibility or implementation of H.265 support within the context of AutoConnect in QGroundControl.

## Log Files and Screenshots
[image]

HTRamsey, 2024-05-18:
Could you enable extra GStreamer debugging in the console and see if it is the same error as here as in #11244? Has to do with the udpsrc buffer size. I wonder if only udp related autoconnect settings are affecting this for you.

nechtobolshee, 2024-05-20:
@HTRamsey Hello! Thank you for your response. 
It seems to me that in this case, the GStreamer logs won't provide much information. 
If you still need to get this information, I can provide it. Here is a more detailed description of the problem I encountered:

1. **pymavlink** doesn't have a native H265 type, so I'm using **mavlink.VIDEO_STREAM_TYPE_RTPUDP**.
[image]

2. As a result, when QGroundControl retrieves the stream information and tries to parse it, the following issue occurs:

In **src/VideoManager/VideoManager.cc** - QGC sets the prefix "udp://" for the H265 stream, which later causes a crash because the H265 stream is attempted to be read as H264
https://github.com/mavlink/qgroundcontrol/blob/af1df0289769c432d60b4a2296db41e8fd9a525b/src/VideoManager/VideoManager.cc#L688

Additional stream info: 
[image]

nechtobolshee, 2024-05-24:
@HTRamsey Hello!
Are there any updates regarding this bug?
Thank you!

HTRamsey, 2024-05-24:
I will look into it deeper over this weekend/next week considering the extra info you provided. GStreamer isn't my expertise so it depends a little bit on if the problem is just QGC setting video up or an actual GStreamer configuration problem. In the worst case scenario, I have another PR that will introduce QtMultimedia as an alternative Video Player (which will use FFMPEG) and that should hopefully resolve the problem with all of the same functionality.