mavlink/qgroundcontrol #11524: QGroundControl build on Jetson Orin Nano failed

lida2003, 2024-05-13:
Is there anything I missed here?

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy

$ tar -xpf qt-everywhere-src-6.6.3.tar.xz
$ cd qt-everywhere-src-6.6.3/
$ ./configure -xcb -skip qtwebengine
```

failed with following log:

```
CMake Error at qtbase/cmake/QtBuildInformation.cmake:523 (message):
  Feature "xcb": Forcing to "ON" breaks its condition:

      QT_FEATURE_thread AND TARGET XCB::XCB AND TEST_xcb_syslibs AND QT_FEATURE_xkbcommon_x11

  Condition values dump:

      QT_FEATURE_thread = "ON"
      TARGET XCB::XCB found
      TEST_xcb_syslibs = "FALSE"
      QT_FEATURE_xkbcommon_x11 not evaluated

Call Stack (most recent call first):
  qtbase/cmake/QtFeature.cmake:317 (qt_configure_add_report_error)
  qtbase/cmake/QtFeature.cmake:437 (qt_feature_check_and_save_internal_value)
  qtbase/cmake/QtFeature.cmake:706 (qt_evaluate_feature)
  qtbase/cmake/QtFeature.cmake:675 (qt_feature_module_end)
  qtbase/src/CMakeLists.txt:13 (qt_feature_evaluate_features)

```

cat "/home/daniel/Work/qt-everywhere-src-6.6.3/CMakeFiles/CMakeError.log"

```
int main(int /*argc*/, char ** /*argv*/)
{

    UaPlatformLayer::init();
    UaSession *session = new UaSession;

    UaPlatformLayer::cleanup();
    return 0;
}
Performing C++ SOURCE FILE Test HAVE_libhybris_egl_server failed with the following output:
Change Dir: /home/daniel/Work/qt-everywhere-src-6.6.3/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/ninja cmTC_0cc59 && [1/2] Building CXX object CMakeFiles/cmTC_0cc59.dir/src.cxx.o
FAILED: CMakeFiles/cmTC_0cc59.dir/src.cxx.o
/usr/lib/ccache/c++ -DHAVE_libhybris_egl_server  -fPIE -std=gnu++17 -o CMakeFiles/cmTC_0cc59.dir/src.cxx.o -c /home/daniel/Work/qt-everywhere-src-6.6.3/CMakeFiles/CMakeTmp/src.cxx
/home/daniel/Work/qt-everywhere-src-6.6.3/CMakeFiles/CMakeTmp/src.cxx:4:10: fatal error: hybris/eglplatformcommon/hybris_nativebufferext.h: No such file or directory
    4 | #include <hybris/eglplatformcommon/hybris_nativebufferext.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

Source file was:

#include <EGL/egl.h>
#include <EGL/eglext.h>
#include <hybris/eglplatformcommon/hybris_nativebufferext.h>

int main(int argc, char **argv)
{
    (void)argc; (void)argv;
    /* BEGIN TEST: */
#ifdef EGL_HYBRIS_native_buffer
return 0;
#else
#error Requires EGL_HYBRIS_native_buffer to be defined
return 1;
#endif
    /* END TEST: */
    return 0;
}

```

HTRamsey, 2024-05-13:
Do a search for: `target_compile_definitions(qmlglsink PRIVATE HAVE_QT_X11)` and comment it out

lida2003, 2024-05-14:
> Do a search for: `target_compile_definitions(qmlglsink PRIVATE HAVE_QT_X11)` and comment it out

I didn't find any line above in 6.6.3 source code.

Does this something to do with [Qmlglsink Build Fixes ](https://github.com/mavlink/qgroundcontrol/commit/94912eebd8b93fc0b6c3fffd79217543bd691a6f)

Maybe I should use latest master code.

HTRamsey, 2024-05-14:
its in qgroundcontrol/cmake/BuildGstQtPlugin.cmake

lida2003, 2024-05-14:
> its in qgroundcontrol/cmake/BuildGstQtPlugin.cmake

It seems [QT6.6.3](https://download.qt.io/archive/qt/6.6/6.6.3/single/qt-everywhere-src-6.6.3.tar.xz) doesn't have this file `BuildGstQtPlugin.cmake`

```
daniel@daniel-nvidia:~/Work/qt-everywhere-src-6.6.3$ ls cmake/
ECMOptionalAddSubdirectory.cmake  QtSynchronizeRepo.cmake  QtTopLevelHelpers.cmake
```

jinglebang, 2024-05-16:
你好, 我使用Qt6.6.3编译环境，   git  master源码  ， 但是，编译运行成功后 标题是  QGC Daily   我不太知道  我编译运行的是不是正确的，看到您这边使用相同的环境 所以想请教一下 我的是否正确
[image]

HTRamsey, 2024-06-13:
closing in favor of duplicate #11614

closed 2024-06-13 by HTRamsey