mavlink/qgroundcontrol #11544: Clear button offline

arnaudjuin, 2024-05-16:
It would be nice to have an option to clear the planning view when QGC is not connected to the vehicle.

smadi0x86, 2024-06-18:
I agree that having an option to clear the planning view when QGroundControl (QGC) is not connected to the vehicle would be a useful enhancement.

To address this issue, I propose adding a Clear button that is always enabled, regardless of the vehicle connection status. This button would allow users to clear all mission items from the planning view, even when QGC is offline.

**Here's a high-level plan for implementing this feature:**

- Add a Clear button to the toolstrip in the planning view.
- Implement a clearPlanningView function that removes all mission items and resets the view.
- Ensure the Clear button triggers this function.

**Files to edit for this features:**

- https://github.com/mavlink/qgroundcontrol/blob/master/src/PlanView/PlanView.qml

I would love to hear your feedback on this approach or any suggestions you might have before I start working on a pull request.

Thanks!

arnaudjuin, 2024-06-19:
That's the way I see it. I can also help you with the PR.

smadi0x86, 2024-07-19:
Hello, I hope you're doing well, Sorry for the delay in working on this, I was on vacation. I've put together a PR (https://github.com/mavlink/qgroundcontrol/pull/11691)  to address the issue and detailed what I've done in it.

Turns out my earlier insights were incorrect after diving deep into the code, the logic to enable the "Clear" button when offline was already in place. I tried tracing and changing the logic to just give a warning when vehicle is offline and execute the `clear` button.

DonLakeFlyer, 2024-07-22:
You can just use File/Blank to start over again. Is that too confusion? Adding a toolstrip button to clear isn't a great use of important real estate. Options like "Clear" are already under the File button.

smadi0x86, 2024-07-22:
I totally agree, in fact I was trying to copy the same functionality/logic of creating a new blank plan for clearing via the clear button. I have no problem with both, if the feature requester have any strong takes on why we need to that, we can possibly continue if that's possible.

DonLakeFlyer, 2024-07-22:
> if the feature requester have any strong takes on why we need to that, we can possibly continue if that's possible.

Or maybe just re-oeganizse/re-name things to make them more clear. We'll see where the. confusion is.

smadi0x86, 2024-07-23:
Yes that's the way I see it, do you have anything in mind?

DonLakeFlyer, 2024-08-01:
I think maybe a solution is to move the Clear out of the Vehicle section so that it's appropriate for use at all times. Let me think about that a bit.

DonLakeFlyer, 2024-08-01:
@arnaudjuin Any comments on why you didn't think to use File/Create Plan/Blank?

mrpollo, 2024-08-01:
Clear Mission / Clear Plan sounds like a positive change to the menu item

closed 2024-08-10 by DonLakeFlyer

arnaudjuin, 2024-08-12:
> Contributor

I am happy with that, it makes more sense.

arnaudjuin, 2024-08-12:
Moving it out of the Vehicle section makes more sense indeed.