mavlink/qgroundcontrol #11554: QGCFileDialog Doesn't Open and seems to Cause a Freeze

HTRamsey, 2024-05-18:
Mostly likely related to same Abstract Button issue as everything else

DonLakeFlyer, 2024-08-20:
@HTRamsey I fixed a bunch of Qt6 stuff with this a while back. Do you still have an example that doesn't work? I tried some on OSX and they work. Is it OS specific?

HTRamsey, 2024-08-20:
Hmm I just checked and the one I remembered having issues with seems working

closed 2024-08-20 by HTRamsey

mrschick, 2025-04-22:
I'm building today's `master` branch on `Ubuntu 24.04.2 LTS` for testing purposes and getting exactly this issue.

When selecting any "open file" prompt I've encountered so far, whether the "3D Map File" selector or "Save App Log" in the console, the UI becomes completely unresponsive, only killing the process and restarting helps.
No related output log is written in the terminal instance I used to launch the QGC build.

mrschick, 2025-04-22:
I'm not sure why it let me build just fine before fixing it, as reverting doesn't allow it to build.

Basically, CMake would not find the Qt6 dependency, leading to the following error when attempting to run `cmake --build build --config Debug`:
```
CMake Error at CMakeLists.txt:162 (find_package):
  By not providing "FindQt6.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt6", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt6" (requested
  version 6.6.3...6.8.3) with any of the following names:

    Qt6Config.cmake
    qt6-config.cmake

  Add the installation prefix of "Qt6" to CMAKE_PREFIX_PATH or set "Qt6_DIR"
  to a directory containing one of the above files.  If "Qt6" provides a
  separate development package or SDK, be sure it has been installed.
```

The way I fixed it was by adding the path to my `Qt6Config.cmake` at `CMakeLists.txt:9`:
```cmake
set(Qt6_DIR ~/Qt/6.8.3/gcc_64/lib/cmake/Qt6)
```

Though since it modifies the checked in `CMakeLists.txt`, I presume there should be a much better system-agnostic solution?

DonLakeFlyer, 2025-04-22:
Something changed some time yesterday with respect to this. I used to be able to build in VS code without these environment vars (Qt6_DIR/CMAKE_PREFIX_PATH). Now I need to set CMAKE_PREFIX_PATH to get the build to work if you aren't using Qt's cmake. That said I don't know why it worked before and not now.

You don't need to modify the build scripts. You need to set either environment variable as the errors says above. I believe you can do it on your cmake command line with a '-D' maybe.

HTRamsey, 2025-04-22:
It's probably the find_program(QMAKE_EXECUTABLE) thing that was moved to prechecks, but is currently commented out because you had an issue with it. That's the only thing that's changed in the past few days not related to deployment

DonLakeFlyer, 2025-04-25:
My suggestion would be to use qt-cmake for the configure step like the docs show: https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/. This removes lots of problems.