mavlink/qgroundcontrol #11556: The ability to undo modifications

flyrobot27, 2024-05-19:
- What problem does it solve?

The ability to undo has been embedded into multiple type of user interactive tools, such as Microsoft Office, IDEs, 3D animation software and design tools such as AutoCAD. It is a bit surprising that QGroundControl does not support these kind of features.

Human makes mistakes, and sometimes we make actions that we didn't mean to. For example, accidentally removed the wrong vertex, removed the wrong waypoing, changed the parameters accidentally, stuff like those.

- What does it do?

When a hot key is pressed (usually ctrl-z), the program will revert the planner to its previous state, before said user modification

- What flight stacks must it work with? (All, PX4, ArduPilot)
N/A
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
N/A
- Are there other systems that have this feature?
N/A
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
N/A

HTRamsey, 2024-05-20:
This is a little tough to implement in practice for most things. What would you say is the most important function to be able to undo?

flyrobot27, 2024-05-20:
Hi there, thank you for the reply. I think right now, the most important function are undoing deletes (e.g. vertex / waypoint deletion). For other stuff (e.g. parameter modification, added waypoint, etc.), I think we can live with those right now.

closed 2026-10-04 by github-actions[bot]