mavlink/qgroundcontrol #11557: [翻译引发的BUG]BUG caused by translation

13022591351, 2024-05-20:
[问题我已经修改编译测试过]I have modified the problem, compiled it and tested it

## Steps to Reproduce:
1.[设置语言为英文去创建任务，结果正常]Set the language to English to create the Misssion, the result is normal
2.[设置语言为中文去创建任务，结果不正常]Set the language to Chinese to create the Misssion, but the result is abnormal.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04.6 LTS
- QGC Version: v4.3.0
- QGC build: daily and stable
- Flight Controller: [与飞控无关]Has nothing to do with flight control
- Autopilot (with version): [与飞控无关]Has nothing to do with flight control

## Detailed Description
[我发现和语言有关后，我修改了translations/qgc_json_zh_CN.ts中和FactComboBox相关的'，'为‘,’编译运行后发现功能正常]After I discovered that it was related to the language, I modified the ',' related to FactComboBox in translations/qgc_json_zh_CN.ts to ',' and found that the function was normal after compiling and running.

YCFPV, 2024-05-29:
大佬

HTRamsey, 2024-05-31:
closing due to #11570 providing solution

closed 2024-05-31 by HTRamsey