mavlink/qgroundcontrol #11559: QGC 4.4 Issues List

rmackay9, 2024-05-20:
This is a consolidated list of issues related to the 4.4 beta release.  Discussions can be found in the AP forum here.

Reports requiring investigation:

- None

Confirmed issues:

- None

Enhancements or Issues that may not be resolved before the stable release:

- None

timtuxworth, 2024-05-22:
I've noticed hangs and crashes on both Mac OSX and Windows 11 when connected to a vehicle (real and SITL) via UDP

vacoff, 2024-05-27:
I came across some other issues. The gimbal icon is missing in all of the v4.4 release candidates. I am using gremsy gimbal with cube orange px4 v1.14

julianoes, 2024-06-02:
@vacoff can you post a screenshot?

@timtuxworth any chance you can give tips how to reproduce the crashes, or get crash reports?

timtuxworth, 2024-06-02:
> @vacoff can you post a screenshot?
> 
> @timtuxworth any chance you can give tips how to reproduce the crashes, or get crash reports?

The only thing special I do is I usually use UDP to connect to my vehicles - this is usually SITL but I also do this with real planes.

It's pretty easy. Just start QGC and use it for a bit. I run SITL for a while doing some testing. It will hang after a while and then you are forced to close it. I make sure to press the "report to Microsoft" button. You should be able to go to your Microsoft dev account and review the crash logs.

Actually the same thing happens on my Mac too. It's very unstable it shouldn't be hard to reproduce.

julianoes, 2024-06-02:
> You should be able to go to your Microsoft dev account and review the crash logs.

No idea how this would all connect. I rarely use Windows.

That being said, I assume QGC v4.3.0 works nicely and doesn't have the stalls and crashes?

timtuxworth, 2024-06-02:
I have no idea. I usually use the daily build which has always worked for me on both windows and Mac.

rmackay9, 2024-06-02:
@timtuxworth @julianoes,

I have not seen QGC 4.4 lockup or crash.  This includes running it on Windows 11 using UDP or USB to real vehicles.

@timtuxworth, could you just confirm that you're using QGC 4.4-rc3?  If the crashing issue involves another version (like 4.5) I'd like it if we could move the discussion to another thread or issue so that we don't muddy the waters of the 4.4 beta testing.

julianoes, 2024-06-02:
@rmackay9 I had a look through the issues you mentioned:

>  Sluggish movement when both PC & Herelink controlling gimbal

This one is tricky. I think it comes down to several points:
- Are both QGC sending with the same sysid/compid? If so, that's not a valid use case, in my opinion.
- If not, how are both sending signals? Commands, or using the wheel/buttons? Basically, the gimbal protocol has the notion of "in control", so in theory it shouldn't really be possible to have two ground stations both controlling the gimbal, unless one is primary and one is in secondary control. The exact behaviour in that case is then implementation specific.
- In the current ArduPilot case, I believe "in control" is not implemented yet, right? If so, that would potentially explain conflicting/sluggish behaviour.
- When I tried this with PX4 (using different sysids for the two QGC instances). I had to manually "release control" in order for the other QGC to "acquire control". Otherwise, it would not work. (A user warning for this case would be a nice to have in the future.) Mixing RC and MAVLink gimbal input did not work, presumably that's a PX4 issue.

> Herelink click-and-drag not working -- reported by HenryW

This would depend on the setup. It worked for me, although very slowly. And I'm not sure if it's the right way around. Could it be inverted? Maybe @Davidsastresas you can clarify: is it supposed to move with my finger? E.g. I point and move my finger the right and expect the camera to pan left?

> "Point Home" ROI can't be cancelled, remains on map -- reported by Randy

I can confirm that. I'm not sure how to fix that one. @Davidsastresas hints would be appreciated.

> "Point Home" appears not to use the correct altitude (camera points up) -- reported by Randy

This seemed correct with my testing with PX4. I'd check what's going on in:
https://github.com/mavlink/qgroundcontrol/blob/Stable_V4.4/src/Vehicle/Vehicle.cc#L2869-L2912

vacoff, 2024-06-04:
> @vacoff can you post a screenshot?
> 
> @timtuxworth any chance you can give tips how to reproduce the crashes, or get crash reports?

this screenshots are from release candidate 4.5 but the issue is still ongoing, even though the gimbal control works ok from both console and the rc input
[image]
[image]

Davidsastresas, 2024-06-04:
- About cancel ROI when "ROI home" action, it can be done. The area clickable is not only the "waypoint icon", but also the label. If you click the label that popup to cancel ROI will appear:
[image]

- About click and drag, I am thinking on disabling it for the moment. In android it is not working as it is supposed to, and I think it is prone to missunderstand how it works. It is basically a virtual joystick occupying the whole screen. I played around with inverting the direction as some comments suggested, but it wasn't natural, it wasn't feeling like when you pan a map, because the camera isn't tracking the position really, is being controlled by speed on that command.

So I think it is better to disable it for now, and come back to it in the future with more time to test it maybe. Any thought on this?

- About the crashes, I can't comment about Macos, but me and my team use this regularly in Linux and Windows 10 and 11, and we haven't seen any crash lately. A few months ago, a crash bug was discovered related to joystick, it would only make QGC crash if a joystick was present and connected, but all these 4.4 release candidates have the patch, as well as master. So we really need more information about exact versions, etc to try to reproduce.

julianoes, 2024-06-04:
@vacoff ah, so the gimbal doesn't get detected in your case. Now I understand. What is the MNT_MODE_IN param set to?

rmackay9, 2024-06-05:
Hi @Davidsastresas,

I think we should leave the click-and-drag in.  I think it works unless there are two ground stations connected at which point they interfere with each other.  I think it does not harm to leave it in.

I vaguely remember (I think) that you said it was implemented using angle control, I think we should change that to use rate-control (which DO_GIMBAL_MANAGER_PITCHYAW) supports but I think we could do that for QGC-4.5.

julianoes, 2024-06-06:
@vacoff I think I have found an edge case in PX4 that causes GIMBAL_MANAGER_INFORMATION not to be sent out and hence QGC not detecting the gimbal. I'll have a fix soon.

julianoes, 2024-06-06:
@vacoff can you test with this PX4 branch: https://github.com/PX4/PX4-Autopilot/pull/23236

vacoff, 2024-06-06:
> @vacoff ah, so the gimbal doesn't get detected in your case. Now I understand. What is the MNT_MODE_IN param set to?

it is set to mavlink v2

vacoff, 2024-06-06:
@julianoes just tested with the pr you said. GImbal icon is now appearing on the QGC but when I try to command it the error on the second picture is appearing. it is v4.4 rc5. It seems like it is stuck on the yaw follow mode and doesn't accept commands to change its mode also the on screen control doesn't work. it reacts to the "gimbal test" commands. 

btw I have changed hardware from cube orange to cuav nora as wouldn't like to fly with un-official firmware just to be safe. But I am not sure if this would make any difference.

[image]
[image]

vacoff, 2024-06-06:
After rebooting the autopilot the gimbal manager command rejection error disappeared, but the whole logic of the commands on QGC is working kinda weird. For gimbal is on yaw follow mode in default. After pressing yaw lock it locks it in place until another command is activated such as center or tilt 90 degrees then it reverts to the yaw follow mode.

julianoes, 2024-06-06:
@vacoff I'm glad you got it to work (somewhat).

> And the yaw lock mode makes gimbal twitch in the yaw axis,

I think that's a gimbal issue. I'm assuming this is using the SIYI A8 mini? If so, I see the same.

> After pressing yaw lock it locks it in place until another command is activated such as center or tilt 90 degrees then it reverts to the yaw follow mode.

That's by design. I don't think locking at 90 degrees down would make sense. And when you click center you want to have it look forward, so it shouldn't be locked anymore either.

vacoff, 2024-06-09:
> I think that's a gimbal issue. I'm assuming this is using the SIYI A8 mini? If so, I see the same.

I am using Gremsy Mio. It seems like gimbal is switching between yaw lock and yaw follow modes fast, so it causes twitching.

julianoes, 2024-06-09:
@vacoff that's interesting. Do you know what's sending it follow commands?

julianoes, 2024-06-09:
@vacoff I verified this and I can't see the effect, at least when using PX4. It either sends with or without yaw lock. There is no switching between the two modes in my testing.

HTRamsey, 2024-07-27:
Can someone else verify the Offline Maps page doesn't work because of missing property `defaultSuffix`?

soleongmbh, 2024-08-05:
I have installed 4.4 on two Herelinks. The app seems to work well so far.
We don't use gimbal control,
If we use only the camera conneted to HDMI all is working fine, but switching from stream 1 (pilot camera) to stream 2 (gimbal camera) doesn't seem to work.
This is very important for us.

julianoes, 2024-08-05:
@soleongmbh thanks for the report. We tested the HDMI functionality, so that's interesting.

@rmackay9 do you remember what steps you took that fixed this?

rmackay9, 2024-08-06:
Hi @soleongmbh,

I wonder if you're sure that switching between the streams worked before upgrading the herelink QGC firmware?

There are instructions here for enabling two streams on the herelink https://docs.cubepilot.org/user-guides/herelink/herelink-faq#herelink-dual-video-instruction

soleongmbh, 2024-08-06:
@julianoes   on Herelink with "old" QGC the switching was working, it was really slow, but it was working. We use as pilot-cam the Hawkeye Firefly 4k split with setting "forced HDMI out" 
and as Gimbal-Cam the Sony A7R 3
@rmackay9  With the "old" QGC, switching between the video streams also worked without activating the dual stream on the Herelink. Is this function necessary for the new QGC?

rmackay9, 2024-08-06:
Hi @soleongmbh,

I think I'll have to re-test but as far as I can remember, I did not actually have to follow the instructions that I linked above in order to enable two HDMI steams. The only caveat is that I think I only ever tested using one HDMI stream at a time.  So I tested moving the same camera between the herelink's HDMI1 and HDMI2 inputs but maybe I never tested two cameras attached at the same time.

soleongmbh, 2024-08-07:
Hi @rmackay9 , let me know if you want me to help with testing. It would be important that the switching between the streams works reliably.

soleongmbh, 2024-08-19:
Hi @rmackay9 , Could you find the problem? Thank you!

julianoes, 2024-09-02:
@soleongmbh did you ever get to the bottom of this? Or is switching still not working?

soleongmbh, 2024-09-03:
Hi @julianoes , Unfortunately, switching between the two video streams is still not working. We have also tried different Herelinks. With the “old” QGC, the switching works, albeit slowly, but it works.

julianoes, 2024-09-03:
These are Herelink 1.1 or 1.0 and updated AirUnits?

soleongmbh, 2024-09-04:
@julianoes It was Herelink 1.0 with updated Airunits.

julianoes, 2024-09-06:
Ok, I just tested the side loaded app using a Herelink 1.0 with Airunit also at 1.0 both updated and I can switch HDMI with the stock QGC app, as well as the v4.4.0 release. Not sure what else to try.

julianoes, 2024-09-06:
@soleongmbh the latest to check would be https://github.com/CubePilot/qgroundcontrol-herelink, just building right now.

soleongmbh, 2024-09-06:
@julianoes 
 Hi, can not find the apk for herelink.

soleongmbh, 2024-09-06:
@julianoes I installed the 4.4.2 version on Herelink. here you can find the information abiout the version of Herelink and Airunit and two videos: 
- test switching videostreams on old QGC
- test switching videostream on QGC 4.4.2
Link: https://www.dropbox.com/scl/fo/313n6ic970w5mgi0q3chh/AH-mqi5BAFdLHItgJNG74JE?rlkey=u5pmbq5ipkoeiznzyo90n96iy&dl=0

julianoes, 2024-09-07:
@soleongmbh thanks for this, so I'm watching 20240906_171354-QGC-4.4.2:

0:52: switch to HDMI1 worked
1:12: switch to HDM2 eventually worked
1:27: switch to HDMI1 worked again
2:05: switch to HDMI2 didn't work, or was too slow
end: switch to HDMI2 didn't work again

I was under the impression that it doesn't work at all for you but it looks like it doesn't work consistently, or maybe only in the beginning? Do I get that right?

soleongmbh, 2024-09-07:
@julianoes Je, you are right. it works a few times at the beginning and then no more. Once it stops working, I have to powercycle the drone and the Herelink to make it work again.

Loading the parameters should not be a problem for my understanding, if the switching works with the old QGC, then it should also work with the new QGC. 

I wonder if the switching works for you several times and without problems?

julianoes, 2024-09-10:
@soleongmbh I think I can reproduce the issue. It seems to get in a state where it is out of sync, thinking it's showing HDMI2 but falling back to HDMI1 somehow. I will have to have a closer look.

soleongmbh, 2024-09-10:
@julianoes Yesterday I tested again with another system where there is no problem with the Mavlink traffic. Also to clarify this doubt.
- With the old QGC everything works normally
- with the new QGC, switching sometimes works and sometimes not. 

Link Videos: https://www.dropbox.com/scl/fo/bt5fgma57dzi7u1gk30lm/AHcKDlsZYWRJ8iuhCsMMoSY?rlkey=m9ar9ek4nu3eclsfus2mgo8n1&dl=0

Davidsastresas, 2024-09-10:
Happy to hear you reproduced it @julianoes . It might be related to the fact that we are not using the feedback from Airunit in QGC 4.4 because we decided to leave that out due to Cubepilot using custom mavlink for this matter, so in QGC 4.4 that is working in "open loop". Is that why you suggested to implement retransmission there?

julianoes, 2024-09-10:
Right, so I'm just wondering if we somehow get out of sync because we don't have feedback and only try once. I'll have to dig deeper. Planning to do that in the next few days.

julianoes, 2024-12-02:
@soleongmbh sorry it took a while, I finally got around and good look closer at the HDMI switching issue. A fix is coming soon: https://github.com/CubePilot/qgroundcontrol-herelink/pull/57

HTRamsey, 2024-12-02:
@julianoes should I look into that fix in master as well? I assume it's relevant to all platforms not just herelink

soleongmbh, 2024-12-02:
@julianoes thank you for the info, if you need help in testing, please let me know if I can help with testing.

julianoes, 2024-12-02:
@DonLakeFlyer this is the current v4.4.3 herelink repo, it's a few commits on top of v4.4.3 upstream:
https://github.com/CubePilot/qgroundcontrol-herelink/tree/Stable_V4.4.3-Herelink

The main thing that can't use custom is switching HDMI, as described here:
https://github.com/DonLakeFlyer/Herelink-QGC-Save/pull/24

rmackay9, 2024-12-03:
@DonLakeFlyer,

BTW, the Herelink is still shipping with a really old version of QGC but I heard from Sid that they had it in their project plans to update it to the latest version.  I'll give them another ping about that.

We have instructions [here on the AP wiki](https://ardupilot.org/copter/docs/common-herelink.html#updating-qgc) about how to update to the 4.4 version.

As a side note I wonder what our plans are for beta testing and releasing the next stable version of QGC based on daily/latest.  Maybe this isn't the place to discuss that but I just want to say that I'm more than happy to help with the beta testing similar to what we did for 4.4

DonLakeFlyer, 2024-12-03:
Regarding the new QGC stable, I think were getting closer to that.

I just ran into a bit of a roadblock on the daily build version of Herelink. My Herelink seems to have crapped out. Do you guys know someone I can ping other than Philip to try to help with resurrecting it? Screen started flashing then it seems to have crashed. And now I can't restart it. I was working on getting that finished up and working with all the new cmake stuff. Still more custom build cmake work to be done there.

DonLakeFlyer, 2024-12-03:
> My Herelink seems to have crapped out.

Nevermind. It seems to have magically resurrected itself after sitting for a couple hours!

rmackay9, 2024-12-03:
@DonLakeFlyer, Great that it's sorted.  In general Sid is the one we normally talk to about hardware support for devs.