mavlink/qgroundcontrol #11564: Video Recording does not work with H265 stream

jnomikos, 2024-05-24:
## Expected Behavior
Recording a H265 stream should work

## Current Behavior
When I have a UDP H265 stream open (no MAVLink camera), and try to record, I get the error "VideoReceiverLog: Failed to link valve and file sink "udp265://0.0.0.0:5600", and the recording does not start. This does not happen with H264 streams. I believe there is some difference in gstreamer flags which is causing this. I have tested this on 4.3, the most recent release ([v4.4.0 release candidate 3](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.4.0rc3)), and the daily build

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Stream a UDP H265 video stream
2. Go to Application Settings -> General. Under "Video Settings", choose "UDP H265 Video Stream" and select the port you are on
3. Attempt to hit the record button when video is selected and view the console output

## System Information
When posting bug reports, include the following information
- Operating System: [Ubuntu 22.04]
- QGC Version: [Tested on 4.4.0, 4.3.0, and daily]
- QGC build: [Most recent release]
- Flight Controller: [Cube Orange Plus]
- Autopilot (with version): [PX4 1.14.0]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

HTRamsey, 2024-05-24:
Seems related with #11523 which also links to other similar issues). I think these video problems all involve the linux version.

jnomikos, 2024-05-24:
> Seems related with #11523 which also links to other similar issues). I think these video problems all involve the linux version.

Yeah I just tested on Windows and I am not seeing the same issue there. 

Also, as a note, my Ubuntu system Gstreamer is on version 1.20.3. 

```
gst-launch-1.0 --gst-version
GStreamer Core Library version 1.20.3
```

jnomikos, 2024-05-24:
Okay, I got video recording working. This was a dependency issue on my end.

The steps I took to get this working.

1. I removed gstreamer like an idiot (this caused my display to go black, I do not recommend doing this)

2. I reinstalled gstreamer and rebooted my pc (same version here, so nothing changed)

3. I double checked the requirements to run QGC and installed them all

`sudo usermod -a -G dialout $USER
sudo apt-get remove modemmanager -y
sudo apt install gstreamer1.0-plugins-bad gstreamer1.0-libav gstreamer1.0-gl -y
sudo apt install libfuse2 -y
sudo apt install libxcb-xinerama0 libxkbcommon-x11-0 libxcb-cursor0 -y`

I think I was missing specifically gstreamer1.0-libav. My bad. I am closing this

closed 2024-05-24 by jnomikos