mavlink/qgroundcontrol #11587: Unable to build custom qgc

zerocool-z1, 2024-06-02:
i am unable to build custom qgc even after following each and every step in qgc developer guide. even the folder doesn't show in the qt, after renaming it to custom. 
[image]

whenever i am trying to build for debugging it is only building the main qgc, not the custom one please help

HTRamsey, 2024-06-02:
Custom Builds are the last part that needs to be updated for CMake.
Check out #11436

zerocool-z1, 2024-06-02:
@HTRamsey can you please help me out on a VC or anydesk for the same?

zerocool-z1, 2024-06-02:
i tried to include custom build definations instead of daily build in cmakelists.txt but got the following error

[image]

HTRamsey, 2024-06-02:
It's not going to work at all right now. CMake doesn't even have the custom build stuff linked. You'd have to use older qmake build in order to use custom plugins, or just directly edit source.

closed 2024-11-30 by HTRamsey