mavlink/qgroundcontrol #11589: Enabling Virtual Joysticks Starts them at 0,0 instead of Centered

HTRamsey, 2024-06-02:
Pretty unsafe.

Unrelated addition,  "qrc:/qml/QGroundControl/Controls/JoystickThumbPad.qml:220:9 Parameter "touchPoints" is not declared". Needs updating to Qt6 Qml parameter style

DieBorr, 2024-06-04:
Ok, i will figure out what is happening, thank you

DieBorr, 2024-06-05:
HI @HTRamsey I have been testing the virtual joystick with both reCenter and nonRecenter settings, but I am not sure what specific issue you are referring to.

This afternoon, I have been working on fixing VirtualJoystick-related issues, including some bugs that occurred when resizing the screen, which could be potentially dangerous.

Could you please describe the bug you are pointing out? I want to ensure I make a comprehensive fix to the VirtualJoystick related-stuff. I think that the bug you are pointing out is the one in the following video, but if it is another one please just explain me.

As soon as you reply, I will make a pull request solving all the bugs I have found related to VirtualJoystick. Thank you.

https://github.com/mavlink/qgroundcontrol/assets/47715559/712042f8-e53a-403b-ba19-d5b61acce1b9

HTRamsey, 2024-06-05:
Yup that's it. I've had the one on the left start at the very top, middle, and bottom which I guess is based on the Auto Center setting.
[image]

Also, I've had the issues shown in #11511 & #11590. Although the second one isn't actually a bug, desktop usage needs something to make it easier to use somehow.

DieBorr, 2024-06-05:
Great. I've fixed the resize issues, ensuring the axis values stay consistent and correcting the startup default value problem. I missed issue #11511 but will handle it later. For now, two bugs are fixed. I'll keep working on it and will update once resolved. Thank you.

PD: Regarding #11590 I think that first we must solve this dangerous problem, then try to figure out how to make virtualJoysticks more UI friendly for desktop users, at least making the user able to takeoff when enabling ReCenter joystick setting.

DieBorr, 2024-06-05:
@HTRamsey take a look, I think that most problems are solved, test it and give me your feedback, I will figure out what to do with #11590 when the pull request is accepted.

HTRamsey, 2024-06-06:
Okay I'll check it out tomorrow. Thanks!

HTRamsey, 2024-06-06:
So it works correctly until you connect to a vehicle, then the thumbpad starts at the corner again. But the other issue of moving off screen due to the video seems to be fixed.

DieBorr, 2024-06-06:
Sorry to hear that, I forget to test it while enable and not enable it, I will try to correct it as soon as possible. At least I'm glad to see that all the other problems are solved. Will try to solve it today at night.

About #4821 (quite a lot time haha) the truth is that now, at least in APM, specifically in ArduCopter, when enabling the Joystick it sends a NaN value which seems to be not admitted by the UAV part so UAV doesn't do nothing until you press each joystick. The reality is that this behavior, at least from the brief glance I have taken, is not done on purpose, please correct me if I'm wrong, I'm not familiarized with virtualJoystick and I just took a quick look to the code for answering you. 
I think that we should not close that Issue/Report at least now. The way I think that we can solve this is like showing both sticks in their right position but sending a NaN that is handled by Vehicle.h / Vehicle.cc and not by the self vehicle , In this way we assure to not having unpredictable behavior. However I don't think this is the right solution as if the vehicle is in the air and the setting is with throttle down it could lead to potential thrust lost as soon as the user touches the left joystick. I would like to see your opinion, I just took a quick look about that and I don't now if I'm forgetting something. 

PD: Sorry for the long message but I think that VirtualJoystick stuff can be potentially dangerous so is important to work together for making it User-safe. I will correct the mistake that you have said as soon as I can, then we should clarify this topic. Really appreciate your time and effort. Thanks!

DieBorr, 2024-06-29:
@HTRamsey remember to mark this as solved 👍.

PR #11604 (2024-07-04): Rework + Fix for VirtualJoystick (#11589 & 11511)
Closes #11589
Closes #11511

This commit solves two issues (#11589 & 11511) that were related. At startup, the VirtualJoystick show random axis values (#11589). Additionally, resizing the screen caused the joystick to malfunction, potentially leading to dangerous behavior (#11511). Now, the values remain stable during a resize, and at startup, the joystick initializes at the middle or bottom, depending on the selected settings. Also added logic for ReCenter settings changed during flight which could make something similar as issue (#11589) (if  [... cut]
closed 2024-07-04 by HTRamsey