mavlink/qgroundcontrol #11602: Crash Caused By Frequent MAVLink Messages

flyingthingsintothings, 2024-06-05:
## Expected Behavior
When QGroundControl receivers frequent MAVLink messages, it should display all of them without issues under the panel that opens by clicking the megaphone (or error sign) in the top-left.

## Current Behavior
When messages occur very frequently (every 100ms for example), QGroundControl freezes after a while. The freeze occurs after around 10 to 20 seconds of receiving these messages.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Create a build of autopilot firmware that outputs lots of messages to the ground control station.
2. Flash it on a flight controller.
3. Connect to it using QGroundControl and wait around 20 seconds.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10 (Version: 22H2; OS Build: 19045.4412)
- QGC Version: 4.3.0
- QGC build: stable
- Flight Controller: Pixhawk 6X
- Autopilot (with version): Custom ArduPilot (latest master branch) with `GCS_SEND_TEXT` somewhere in a loop

## Detailed Description
\/

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
Tried to but couldn't get it to work.
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.
[image]

HTRamsey, 2024-06-05:
On which link type?

flyingthingsintothings, 2024-06-06:
Pixhawk 6X connected over USB to the laptop running QGroundControl. I don't have a telemetry connection set up for wireless communication as it is for testing and development only (but I do have one available for testing if necessary).

HTRamsey, 2024-06-06:
I tried turning a ton of messages to 100Hz without any problem so I think this is most likely specific to GCS_SEND_TEXT which sends a MAVLINK_MSG_ID_STATUSTEXT which can be a costly message to send. I'll need to set up a lua script to send that message specifically to see, then I'll get back to you.

Louis-max-H, 2025-03-13:
Hi,

I have made a Lua script has suggesting HTRamsey, and let in run connected to GQC 4.4.3
With 50 ms delay between message and after 2 min, QGC don't crash anymore. The console menu is a little laggy but nothing special.

Here is the Lua script.

function update ()
  gcs:send_text( 7,  "Hello world")
  return update, 50 -- request "update" to be rerun again 50 milliseconds from now

end
return update, 50

I think it has been fix somehow.

Louis-max-H, 2025-04-07:
I think you can close this issue @HTRamsey , I don't have the permissions to do so

closed 2025-04-07 by HTRamsey