mavlink/qgroundcontrol #11610: Selecting 'MapBox' Map displays blank Map

ashokbugude, 2024-06-11:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open Application settings
2. Navigate to miscellaneous settings 
3. Click on Map Provider dropdown
4.Select MapBox

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14] - Sonoma 14.5
- QGC Version: [e.g. 3.4.4] 
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.] 
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1] 

## Detailed Description
Provide further details about your issue/bug.
Selecting 'MapBox' Map displays blank Map as seen in scrrenshots

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
[image]

- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.
[image]

Blank map
[image]

HTRamsey, 2024-06-11:
@bkueng this would be related to my stuff. Previously it showed Bing as a back up if the API key was not entered or incorrect which seemed kinda misleading to me. Should I re-add that?

ashokbugude, 2024-06-12:
@HTRamsey @bkueng @DonLakeFlyer I request you to kindly let me the location/screenshot where I am supposed to add the mapbox key as I could not find it.

bkueng, 2024-06-13:
> Also @bkueng @DonLakeFlyer this reminds me of something related to my recent QtLocationPlugin stuff. Previously only VWorld showed Bing as a back up if the API key was not entered or incorrect which seemed kinda misleading to me. Should I re-add that? Seems like there should either be no backup or a common one, otherwise the user has no idea what maps they are actually using.

Personally I would not add a backup with the argument that if the user wanted a custom map, it's confusing if some other map is shown. Or at least show an error message.

ashokbugude, 2024-06-13:
Able to see after zooming in more

closed 2024-06-13 by ashokbugude