mavlink/qgroundcontrol #11611: includes for GimbalController.h are missing

christianrauch, 2024-06-11:
## Expected Behavior
Project should build.

## Current Behavior
The 4.4.0 build fails with:
```
/run/build/QGroundControl/src/Joystick/Joystick.cc:19:10: fatal error: GimbalController.h: No such file or directory
   19 | #include "GimbalController.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

## Steps to Reproduce:
build the flatpak
1. clone https://github.com/flathub/org.mavlink.qgroundcontrol
2. build: `flatpak-builder build-dir org.mavlink.qgroundcontrol.yml`

## System Information
When posting bug reports, include the following information
- Operating System: flatpak
- QGC Version: 4.4.0

## Log Files and Screenshots
https://buildbot.flathub.org/#/builders/25/builds/15748/steps/7/logs/stdio

HTRamsey, 2024-06-11:
@julianoes if you have any comments on this you'd be better than me. But I'd say anything before v5.0 should be built with qmake.

julianoes, 2024-06-14:
I usually built the Stable_V4.4 branch using qmake, so I'd be surprised if this is broken. Can you provide steps to reproduce this?

christianrauch, 2024-06-14:
> I usually built the Stable_V4.4 branch using qmake, so I'd be surprised if this is broken. Can you provide steps to reproduce this?

I guess I should have mentioned that I am building with CMake + Ninja. Do the flatpak steps above work for you? That should be the more deterministic way to build it, compared to building it locally in your host environment.

DonLakeFlyer, 2024-06-14:
flatpack isn't something that is actively maintained. So could easily be broken/out of date

christianrauch, 2024-06-14:
> flatpack isn't something that is actively maintained. So could easily be broken/out of date

__I am the maintainer of the `org.mavlink.qgroundcontrol` flatpak.__ There is actually not much to maintain. The build instructions (https://github.com/flathub/org.mavlink.qgroundcontrol/blob/master/org.mavlink.qgroundcontrol.yml) simply run CMake. I basically just need to update to a release tag for new releases and fix any build issues. I am currently updating (maintaining) the instructions to use the newer released version 4.4. But this breaks in a reproducible manner and that is the whole reason why I opened the issue :-)

christianrauch, 2024-06-14:
If you check the log above, you can see that the includes:
```
-I/run/build/QGroundControl/src/ui
-I/run/build/QGroundControl/_flatpak_build/src/ui
-I/run/build/QGroundControl/_flatpak_build/src/ui/ui_autogen/include
-I/run/build/QGroundControl/_flatpak_build/src/ui/ui_autogen/include_Debug
-I/run/build/QGroundControl/_flatpak_build/src/qgc_autogen/include
-I/run/build/QGroundControl/_flatpak_build/src/qgc_autogen/include_Debug
-I/run/build/QGroundControl/src/ADSB
-I/run/build/QGroundControl/src/AnalyzeView
-I/run/build/QGroundControl/src/api
-I/run/build/QGroundControl/src/Audio
-I/run/build/QGroundControl/src/AutoPilotPlugins
-I/run/build/QGroundControl/src/AutoPilotPlugins/APM
-I/run/build/QGroundControl/src/AutoPilotPlugins/Common
-I/run/build/QGroundControl/src/AutoPilotPlugins/PX4
-I/run/build/QGroundControl/src/Camera
-I/run/build/QGroundControl/src/comm
-I/run/build/QGroundControl/libs/qmdnsengine/src/include
-I/run/build/QGroundControl/_flatpak_build/libs/qmdnsengine/src
-I/run/build/QGroundControl/src/Compression
-I/run/build/QGroundControl/src/FactSystem
-I/run/build/QGroundControl/src/FactSystem/FactControls
-I/run/build/QGroundControl/src/FirmwarePlugin
-I/run/build/QGroundControl/src/FirmwarePlugin/APM
-I/run/build/QGroundControl/src/FlightMap
-I/run/build/QGroundControl/src/FlightMap/Widgets
-I/run/build/QGroundControl/src/FollowMe
-I/run/build/QGroundControl/src/Geo
-I/run/build/QGroundControl/src/GPS
-I/run/build/QGroundControl/src/MissionManager
-I/run/build/QGroundControl/src/PositionManager
-I/run/build/QGroundControl/src/QmlControls
-I/run/build/QGroundControl/src/QtLocationPlugin
-I/run/build/QGroundControl/src/QtLocationPlugin/QMLControl
-I/run/build/QGroundControl/src/Settings
-I/run/build/QGroundControl/src/Terrain
-I/run/build/QGroundControl/src/uas
-I/run/build/QGroundControl/src/Vehicle
-I/run/build/QGroundControl/src/VehicleSetup
-I/run/build/QGroundControl/src/VideoReceiver
```
do not contain `src/Gimbal`, which contains the header file `GimbalController.h`.

DonLakeFlyer, 2024-06-14:
Then you are running into the problem that prior to the current daily cmake is not a verified build system. It sort of works if people keep it up to date. But it falls out of date quickly if people don't update both he qmake and cmake systems. It's not requirement to keep it working when you submit a pull. This all changes in daily where we officially move to cmake and get rid of qmake.

julianoes, 2024-06-15:
@christianrauch it sounds like you have determined the fix required. Do you mind making a pull request against the Stable_V4.4 branch? Then we can (at some point) release it with v4.4.1.

christianrauch, 2024-06-15:
It seems there are more includes missing. When compiling locally with Qt 5.12 from the official online installer, I get:
```
[...]/src/QtLocationPlugin/MapProvider.h:18:10: fatal error: QtLocation/private/qgeomaptype_p.h: No such file or directory
   18 | #include <QtLocation/private/qgeomaptype_p.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

closed 2024-11-30 by HTRamsey