mavlink/qgroundcontrol #11614: How to sucessfully build QGC on Jetson Orin Nano?

lida2003, 2024-06-12:
Does anyone have sucessfully build QGC on Jetson Orin Nano?

HTRamsey, 2024-06-13:
Embedded Linux was not an intended target of QGC. That being said, it would likely work if you remove video related stuff.

lida2003, 2024-06-28:
QT 6.3.3 installed, but still got error.

```
/home/daniel/QGC/qgroundcontrol/libs/qmlglsink/qt6/gstqt6glutility.cc
/home/daniel/QGC/qgroundcontrol/libs/qmlglsink/qt6/gstqt6glutility.cc: In function ‘GstGLDisplay* gst_qml6_get_gl_display(gboolean)’:
/home/daniel/QGC/qgroundcontrol/libs/qmlglsink/qt6/gstqt6glutility.cc:94:62: error: ‘QX11Application’ is not a member of ‘QNativeInterface’; did you mean ‘QWaylandApplication’?
   94 |     auto x11_native = app->nativeInterface<QNativeInterface::QX11Application>();
      |                                                              ^~~~~~~~~~~~~~~
      |                                                              QWaylandApplication
/home/daniel/QGC/qgroundcontrol/libs/qmlglsink/qt6/gstqt6glutility.cc:94:79: error: no matching function for call to ‘QGuiApplication::nativeInterface<<expression error> >()’
   94 |     auto x11_native = app->nativeInterface<QNativeInterface::QX11Application>();
      |                                                                               ^
In file included from /home/daniel/QGC/qt6/qt6-install/include/QtCore/qcoreapplication.h:16,
                 from /home/daniel/QGC/qt6/qt6-install/include/QtGui/qguiapplication.h:8,
                 from /home/daniel/QGC/qt6/qt6-install/include/QtGui/QGuiApplication:1,
                 from /home/daniel/QGC/qgroundcontrol/libs/qmlglsink/qt6/gstqt6glutility.cc:26:
/home/daniel/QGC/qt6/qt6-install/include/QtGui/qguiapplication.h:142:5: note: candidate: ‘template<class NativeInterface, class TypeInfo, class BaseType, typename std::enable_if<TypeInfo::isCompatibleWith<QGuiApplication>, bool>::type <anonymous> > NativeInterface* QGuiApplication::nativeInterface() const’
  142 |     QT_DECLARE_NATIVE_INTERFACE_ACCESSOR(QGuiApplication)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/daniel/QGC/qt6/qt6-install/include/QtGui/qguiapplication.h:142:5: note:   template argument deduction/substitution failed:
/home/daniel/QGC/qgroundcontrol/libs/qmlglsink/qt6/gstqt6glutility.cc:94:79: error: template argument 1 is invalid
   94 |     auto x11_native = app->nativeInterface<QNativeInterface::QX11Application>();
      |                                                                               ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-unused-private-field’
[56/612] Building CXX object test/UI/preferences/CMakeFiles/UIPreferencesQmlTestplugin.dir/UIPreferencesQmlTestplugin_uipreferencesqmltestPlugin.cpp.o
ninja: build stopped: subcommand failed.
```

jonazti07, 2024-10-01:
I'm also trying to install QGroundControl in a Jetson AGX Orin.

Any update on this disccusion?

lida2003, 2024-11-18:
There is No 6.6.3 in QT install script. It only support  6.7.0 6.7.1 6.7.2 6.7.3 6.8.0

lida2003, 2024-11-20:
Build on jetson orin + QT 6.7.3 https://github.com/mavlink/qgroundcontrol/issues/12109

closed 2024-11-20 by lida2003