mavlink/qgroundcontrol #11621: Automatic Base Station Setup Broken for Septentrio Receivers

flyingthingsintothings, 2024-06-17:
## Expected Behavior

When a Septentrio GNSS receiver is connected to the device, QGroundControl should automatically set it up and forward RTCM corrections over MAVLink to the autopilot vehicle. The setup should happen in a reasonable amount of time, like half a minute.

## Current Behavior

QGroundControl detects that a Septentrio receiver is connected as the RTK icon is displayed in the status bar, but it fails to set up the receiver correctly and never sends corrections to the autopilot vehicle.

## Steps to Reproduce:

1. Enable "RTK GPS" under "Application Settings > AutoConnect to the following devices"
2. Connect a Septentrio GNSS receiver over USB
3. Wait for the RTK icon to appear in the status bar

## System Information

- Operating System: Windows 10
- QGC Version: 4.4.0
- QGC build: stable
- Flight Controller: Pixhawk 6X
- Autopilot (with version): PX4 1.15.0-beta1

## Detailed Description

I also tried `v4.3.0` and there it works. I know that two pull requests were merged into the GPSDrivers library recently to try and improve automatic setup of Septentrio GNSS receivers as base station. Only one of the two made it into `v4.4.0`.

First pull request (included in `v4.4.0`): https://github.com/PX4/PX4-GPSDrivers/pull/155/files
Second pull request by me (not yet included in `v4.4.0`): https://github.com/PX4/PX4-GPSDrivers/pull/157

## Log Files and Screenshots

- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)

\/

- Autopilot logs when available (post a link).

\/

- Screenshots of QGC to help identify the current issue/bug behavior.

\/

FYI @chiara-septentrio

chiara-septentrio, 2024-07-05:
This would probably be fixed by updating the GPS drivers submodule to one of the most recent commit containing the fix (either a41210e or 0b79ec4 should work). I saw you updated the submodule before stable v4.4 @julianoes, could it be possible to do the same for future release ?

julianoes, 2024-07-07:
I thought I had backported the bugfixes. If not I will have to do so and create v4.4.1 with it.

chiara-septentrio, 2024-07-29:
Would it be possible to have it in daily build or 4.4.1 to make the fix available @julianoes ? Or does some testing need to be done first?

chiara-septentrio, 2024-08-13:
I saw the latest release with the fix, it seems to work fine now. Would it be possible to have the same done for master to ensure the fix is in all future releases?

julianoes, 2024-08-13:
@chiara-septentrio thanks for the reminder. This should do it: https://github.com/mavlink/qgroundcontrol/pull/11771.

PR #11771 (2024-08-14): GPS: update driver submodule
This should fix an issue with the automatic base station setup for Septentrio receivers.

Closes #11621
closed 2024-08-14 by DonLakeFlyer