mavlink/qgroundcontrol #11644: Container build failure: libunwind and other libraries

bendyna, 2024-06-29:
## Expected Behavior
success build using doc https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/container.html

## Current Behavior
build fails, see logs below

## Steps to Reproduce:
1. `docker build  --file ./deploy/docker/Dockerfile-build-ubuntu -t qgc-ubuntu-docker --progress=plain --no-cache .  >& docker.log`
2. `sudo rm -rf build/ && mkdir build && docker run --cap-add SYS_ADMIN --device /dev/fuse --rm -v ${PWD}:/project/source -v ${PWD}/build:/project/build qgc-ubuntu-docker >& build.log`
3. 

## System Information
- Operating System: Ubuntu 22.04, x86_64
- QGC Version: github master
- QGC build: container build

## Log Files and Screenshots
- docker image log 
[docker.log](https://github.com/user-attachments/files/16039514/docker.log)
- build log 
[build.log](https://github.com/user-attachments/files/16039517/build.log)
- CMakeError 
[CMakeError.log](https://github.com/user-attachments/files/16039520/CMakeError.log)

I believe libunwind is not enough, libunwind-dev should be added to docker image

gc-oc, 2024-07-03:
I'm also on Ubuntu 22.04, and at least from what I can tell from your build log, I was running into the same issue. I was able to fix it by adding 
```bash
# Fix Git error
RUN git config --global --add safe.directory /project/source
```
to the Dockerfile just before the cmake command is executed. After doing that, I was able to successfully complete the CMake build until the creation of the app image, where I got the following errors

```bash
-- Creating AppImage
fuse: mount failed: Permission denied
open dir error: No such file or directory
```
When running the docker container with the --privileged flag instead of --cap-add SYS_ADMIN, I seem to get further and get the new error message:
```bash
-- Creating AppImage
linuxdeploy version 1-alpha (git commit ID a29b9a4), GitHub actions build 269 built on 2024-07-01 01:15:42 UTC

-- Creating basic AppDir structure -- 
Creating directory /project/build/AppDir/usr/bin/ 
Creating directory /project/build/AppDir/usr/lib/ 
Creating directory /project/build/AppDir/usr/share/applications/ 
Creating directory /project/build/AppDir/usr/share/icons/hicolor/ 
Creating directory /project/build/AppDir/usr/share/icons/hicolor/16x16/apps/ 
Creating directory /project/build/AppDir/usr/share/icons/hicolor/32x32/apps/ 
Creating directory /project/build/AppDir/usr/share/icons/hicolor/64x64/apps/ 
Creating directory /project/build/AppDir/usr/share/icons/hicolor/128x128/apps/ 
Creating directory /project/build/AppDir/usr/share/icons/hicolor/256x256/apps/ 
Creating directory /project/build/AppDir/usr/share/icons/hicolor/scalable/apps/ 

-- Deploying dependencies for existing files in AppDir -- 

-- Copying files into AppDir -- 

-- Copying files into AppDir -- 

-- Deploying files into AppDir root directory -- 
WARNING: Could not find desktop file in AppDir, cannot create links for AppRun, desktop file and icon in AppDir root 

-- Running output plugin: appimage -- 
[appimage/stdout] Found appimagetool: /tmp/.mount_linuxduGoBwa/usr/bin/appimagetool
[appimage/stderr] Running command: /tmp/.mount_linuxduGoBwa/usr/bin/appimagetool "/project/build/AppDir"
[appimage/stderr] 
[appimage/stdout] 
[appimage/stderr] appimagetool, continuous build (commit 5735cc5), build <local dev build> built on 2023-03-08 22:52:04 UTC
[appimage/stdout] WARNING: appstreamcli command is missing, please install it if you want to use AppStream metadata
[appimage/stderr] Desktop file not found, aborting
[appimage/stdout] WARNING: gpg2 or gpg command is missing, please install it if you want to create digital signatures
ERROR: Failed to run plugin: appimage (exit code: 1) 
```
I'm not sure how the desktop file should be copied into the /project/build/AppDir directory, but it's not being copied there.

Even if I go as far as to specify the executable, icon, and desktop file locations inside the CreateAppImage.cmake file itself, an AppImage will be created that just crashes immediately.
```cmake
# Modification to line 21
execute_process(COMMAND ${LD_PATH}
    --appdir ${APPDIR_PATH}
    --executable ${CMAKE_SOURCE_DIR}/staging/bin/QGroundControl
    --icon-file ${CMAKE_SOURCE_DIR}/staging/share/icons/hicolor/128x128/apps/org.mavlink.qgroundcontrol.png
    --desktop-file ${CMAKE_SOURCE_DIR}/staging/share/applications/org.mavlink.qgroundcontrol.desktop
    --output appimage) # --plugin qt
```
Below is the result of running the AppImage.
```bash
$ ./QGroundControl-x86_64.AppImage 
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Aborted (core dumped)
```
Not super sure where to go from here.

build log:
[build.log](https://github.com/user-attachments/files/16089563/build.log)
docker log:
[docker.log](https://github.com/user-attachments/files/16089564/docker.log)

HTRamsey, 2024-07-04:
Hmm I just built it without any problem, but I think the docs aren't updated.

gc-oc, 2024-07-04:
Just tried to build on a different machine, also with Ubuntu 22.04, and it succeeded for me as well. I'll update when I figure out what was happening on the first machine.

bendyna, 2024-07-05:
> Hmm I just built it without any problem, but I think the docs aren't updated.

Could you hep me debug and check:
- in your docker image if <sys/sysctl.h> header exists
- this line in cmake build "Looking for sysctlbyname ..."
- Source code in "Determining if the sysctlbyname exist failed with the following output:" in CMakeOutput.log

gc-oc, 2024-07-05:
Alright, I figured out exactly what was going wrong. I was following the instructions laid out [here](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/custom_build/create_repos.html), and imported the repository two days ago. At the time, the most recent commit was [5a348d3](https://github.com/gc-oc/qgroundcontrol_custom/commit/5a348d3cae2e6c9781fdb39b9b1b4edadfd24c80). In that version of the repo, the logic in the CMakeLists.txt file that sets the LINUX variable to TRUE 

```
87 if(UNIX AND NOT APPLE AND NOT ANDROID)
88     set(LINUX TRUE)
89 endif()
```

comes after the first time the LINUX variable is referenced:

```
40 if(LINUX)
41     set(CMAKE_INSTALL_PREFIX ${CMAKE_BINARY_DIR}/AppDir/usr)
42 else()
43     set(CMAKE_INSTALL_PREFIX ${CMAKE_BINARY_DIR}/staging)
44 endif()
```

Which obviously leads to the CMAKE_INSTALL_PREFIX being wrong. This specific issue was fixed yesterday by [d19e4c9](https://github.com/mavlink/qgroundcontrol/commit/d19e4c94ec2d55e911d71bc98077b2277fb75062). After swapping the order of these statements, I can build fine.

HTRamsey, 2024-07-05:
Oh okay good, I am also including a few more libraries like appstream, libunwind, and gnupg in a new PR

closed 2024-07-23 by HTRamsey