mavlink/qgroundcontrol #11647: Appilication Upgrade issue

A908gao, 2024-07-02:
Hi,I'm a developer of SIYI Tech. Last week, I submitted a request for online upgrade support for the PX4 firmware of the N7 flight controller. I would like to inquire whether there has been an update to the application download on the QGroundControl official website. I have noticed updates to the code on GitHub, but the application on the QGC official website still does not support the N7 flight controller.Could you please check this? Thanks

DieBorr, 2024-07-05:
Hi, as I suppose Daily and Stable Qgc application are not submitted to the website every day. I'm not part of the developers that update the website but as far as I know this process can take some time ( I just don't really know how much time ) you can ask this through qgroundcontrol discord so they can answer your question.
If you need that support as fast as possible you can run and make a custom executable of qgc app, however there can be some bugs, so be aware of that.

DonLakeFlyer, 2024-07-05:
New daily builds are pushed up with every merged pull requests.

> Last week, I submitted a request for online upgrade support for the PX4 firmware of the N7 flight controller.

Do you mean you submitted a pull request against QGC with the change? if so, point me at it.

A908gao, 2024-07-15:
> New daily builds are pushed up with every merged pull requests.
> 
> > Last week, I submitted a request for online upgrade support for the PX4 firmware of the N7 flight controller.
> 
> Do you mean you submitted a pull request against QGC with the change? if so, point me at it.
https://github.com/mavlink/qgroundcontrol/pull/11626   Here's the pull request i've uploaded ,when I compile locally, after adding id1123 to FirmwareUpgradeController.cc, I can support online firmware upgrades for PX4, but it is still not supported in the QGroundControl daily builds. What else do I need to do to support my 1123 flight controller ID in the daily builds?

closed 2024-10-16 by HTRamsey