mavlink/qgroundcontrol #11675: Connection to vehicle with new ELRS Mavlink-rc

xznhj8129, 2024-07-11:
## Expected Behavior
Using the new Mavlink-rc feature of ELRS and newest INAV development branch https://github.com/iNavFlight/inav/tree/mmosca-mavlinkrc, Mavlink can now be used to control INAV flight controllers and return mavlink telemetry. When using mavproxy or APM, the GCS connect successfully to the vehicle and receives telemetry.

## Current Behavior
QGC, when given the proper connection (/dev/ttyUSB0, baud rate 460800 as specified and verified working with mavproxy), does not connect, as if it did not see the connection or vehicle as valid.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Install https://github.com/iNavFlight/inav/tree/mmosca-mavlinkrc branch to a flight controller
2. Set up ELRS Mavlink-RC connection as specified here https://github.com/ExpressLRS/ExpressLRS/pull/2753
3. Verify link is working using MP, APM or mavproxy
4. Set up same connection in QGC, try to connect

## System Information
When posting bug reports, include the following information
- OS: Linux Mint

- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: 4.3
- QGC build: stable
- Flight Controller: Speedybee F405 Wing Mini
- Autopilot (with version): INAV 8.0, see above

## Detailed Description
Very experimental edge case but once solved could unlock additional capabilities for non-AP/PX4 UAV

xznhj8129, 2024-07-27:
The reason it's not working is that QGC expects MavComponentID to be MAV_COMP_ID_AUTOPILOT1, whereas in Inav:
static uint8_t mavComponentId = MAV_COMP_ID_SYSTEM_CONTROL;

khikmatovfarkhod, 2024-08-22:
hello can we solve this issue please. i still cant connect px4 to qgc but the mavlink messages are showing up on my transmitter.

xznhj8129, 2024-09-03:
issue was fixed with further work on INAV mavlink telemetry implementation

closed 2024-09-03 by xznhj8129