mavlink/qgroundcontrol #11676: Android GStreamer pipeline only starts once at application startup

Frequency21, 2024-07-12:
Disclaimer: This is more probably a GStreamer bug than a QGroundControl. But I'm hoping others ran into this issue and can share their experience or shred some light how to resolve this.

## Expected Behavior
Video stream should start every time, when RTSP URL changes or stream type changes.

## Current Behavior
When QGC is preconfigured with an IP camera's RTSP URL, then it can start displaying it. But after that, if there's a change in the URL, or chaning the video source to disabled then RTSP again, then the display is blank (fully black).

This is occurs only with a specific camera. [Topotek KHP290G609L11](https://topotek.com/KHP290G609L11-1080P-Day-light-640-Thermal1100m-LRF-Triple-Light-Gimbal-HDMIIP-output-p4777857.html)

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Install fresh QGC v4.3.0 or v4.4.0 on an android device (SkyDroid H16 in my case)
2. Set the video source to RTSP at general settings and configure the RTSP URL as well.
3. Exit the settings.
4. Video display is blank.

## System Information
When posting bug reports, include the following information
- Operating System: Android 7.1.2
- QGC Version: 4.4.0
- QGC build: stable
- Flight Controller: Matek H743
- Autopilot (with version): ArduCopter v4.5.0

## Detailed Description
I also tried to change HW decoder to software decoder (it renders a couple of frame every time the watchdog restarts the pipeline, but then it freezes). Increase the rank of another HW decoder (omx). Change the RTSP stream codec from H265 to H264.

For me (a total gstreamer newbie) it looks like the decoder can not negotiate with the upstream elements (the parser elements inside decodebin3, or if I replace the decodebin3 with the HW decoder element, then it cannot negotiate with the parsebin element). This is because parserbin and decodebin3 uses a typefind GSTElement, which guesses the stream format wrong.

Preconfigured startup (see attached logs.txt at line 2652):
> [I] at ../plugins/elements/gsttypefindelement.c:181 - "<typefind> found caps video/x-h265, width=(int)1920, height=(int)1080, framerate=(fraction)0/1, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)main, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, colorimetry=(string)1:3:5:1, tier=(string)main, level=(string)4.1, probability=100"

After URL changes see how the width, height and profile are incorrect (line 7108):
> [I] at ../plugins/elements/gsttypefindelement.c:181 - "<typefind> found caps video/x-h265, width=(int)16, height=(int)16, framerate=(fraction)0/1, parsed=(boolean)true, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)main-10, probability=100"

The video decoder also changes profile (line 2974):
> [I] at ../gst-libs/gst/video/gstvideodecoder.c:1448 - "<amcvideodec-omxarmvideov5xxdecoder0> upstream tags: taglist, video-codec=(string)"H.265\ \(Main\ Profile\)";"

After (line 7259):
> [I] at ../gst-libs/gst/video/gstvideodecoder.c:1448 - "<amcvideodec-omxarmvideov5xxdecoder1> upstream tags: taglist, video-codec=(string)"H.265\ \(Main\ 10\ Profile\)";"

## Log Files and Screenshots
- Log file: [logs.txt](https://github.com/user-attachments/files/16193286/logs.txt)
- Pipelines after videosink is linked.
  - When video starts correctly on application startup: [image]
  - When URL is changed and decoding stops: [image]
- Please notice how similar the two pipeline except inside decodebin3 capsfilter element.
- I also found a similar issue [#9440](https://github.com/mavlink/qgroundcontrol/issues/9440#issuecomment-787417912) but that case the reporter couldn't retest the application with the same camera.

Viliuks, 2025-04-15:
@Frequency21 any updates on this issue? I'm encountering the same thing but on android 9, qgc 4.4.3, gstreamer 1.18.6

HTRamsey, 2025-09-29:
No, it's definitely something I should prioritize but I just haven't yet. I know Don is planning to make an effort at video stuff sometime soon and hopefully I'll join in and fix some of this stuff by then

DonLakeFlyer, 2026-07-29:
Using daily I switched from an RTSP video stream and then to a UDP video stream and it worked fine. Re-open if you can reproduce on latest daily.