mavlink/qgroundcontrol #11679: Daily Android Build Link Returns Access Denied

ericjohnson97, 2024-07-14:
I would like to install the daily build for android.  The link to the following link for the x64 is listed, but returns access denied. 

link: 
https://d176tv9ibo4jno.cloudfront.net/builds/master/QGroundControl64.apk

docs page: 
https://docs.qgroundcontrol.com/Stable_V4.3/en/qgc-user-guide/releases/daily_builds.html

HTRamsey, 2024-07-14:
I don't think the daily on the play store has been updated in a while. Best to download from this repo

ericjohnson97, 2024-07-14:
can you get daily builds from this repo, or only releases?

ericjohnson97, 2024-07-15:
looks like you can get them from the latest android github action 

https://github.com/mavlink/qgroundcontrol/actions/runs/9929355840

DonLakeFlyer, 2024-07-19:
@mrpollo Any ideas here? The APKs are up on S3 but they don't seem to be on cloud front which the website link refers to.

mrpollo, 2024-07-19:
This isnt a CloudFront issue, file names changed and docs didnt get updated.

This is what we have in S3. The docs are linking to the wrong file for linux, and we no longer carry 32 vs 64 bit Android images.

This link should work for the android daily downloads
https://d176tv9ibo4jno.cloudfront.net/builds/master/QGroundControl.apk

[image]

HTRamsey, 2024-07-23:
Fixed by #11690

closed 2024-07-25 by HTRamsey