mavlink/qgroundcontrol #11686: GimbalController uses vehicle compId instead of gimbal compId

FrankvVeelen, 2024-07-19:
## Expected Behavior
All gimbal communication should be with the gimbal manager, hence should use gimbal component id (154 for gimbal 1)

## Current Behavior
Currently _vehicle->compId is used. This results in messages such as acquire control getting lost if the gimbal manager is not on the autopilot.
https://github.com/mavlink/qgroundcontrol/blob/08d6c050c2c22f9dea15944afa86fa0fc93544fc/src/Gimbal/GimbalController.cc#L577-L594

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Attach gimbal with integrated manager
2. Try acquiring control via QGC
3. 
## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04
- QGC Version: 4.4.0
- QGC build: Stable
- Flight Controller: -
- Autopilot (with version): PX4 1.14

julianoes, 2024-07-24:
Thanks for the issue! I disagree that it should be the gimbal component 154 explicitly but you are correct that it could be any component and not necessarily the autopilot.

Do you already have a fix locally and want to contribute a PR? Otherwise, I'll try to look into this at some point, likely next week, and include it in v4.4.1.

FrankvVeelen, 2024-07-25:
I have a local fix where I hardcoded id=154, but I agree that it shouldn't be hardcoded at all. Unfortunately my understanding of the QGC source code is not that deep, but I can take a look. How would you propose to decide the component ID to use? Base it on the heartbeat with MAV_TYPE_GIMBAL with a fallback to the autopilot?

julianoes, 2024-07-25:
I'm working on a fix. I'll let you know when you can test it.

julianoes, 2024-07-26:
@FrankvVeelen I gave this a try. It would be great if you could test https://github.com/mavlink/qgroundcontrol/pull/11706.

FrankvVeelen, 2024-07-26:
Yep, that works for me! Thanks for the quick patch! I can not test if it still works for autopilot-controlled gimbal managers tho.

julianoes, 2024-08-04:
Should be fixed with v4.4.1.

PR #11706 (2024-08-02): [v4.4] Gimbal fixes
This includes several gimbal fixes. For more information, please check the individual commit descriptions.

Fixes #11686.
closed 2024-08-04 by julianoes