mavlink/qgroundcontrol #11687: Virtual joystick extends beyond screen boundaries, preventing throttle maximization

demirciAhmet, 2024-07-19:
## Expected Behavior
The virtual joystick should remain within the screen boundaries at all times. Users should be able to maximize the throttle without the joystick extending beyond the screen.

## Current Behavior
When the virtual joystick is moved towards the edges of the screen, it extends beyond the screen boundaries. As a result, users are unable to maximize the throttle.

## System Information
- Operating System: Pop_OS! 22.04
- QGC build:  self-built from master

## Log Files and Screenshots
[image]

## Possible Fix
The joystick should not re-center when it reaches the edge of the screen. Instead, it should stop at the screen boundary, allowing users to maximize the throttle without losing control.

HTRamsey, 2024-07-19:
@DieBorr Would you want to look at this? I think this is what I noticed in #11590

DieBorr, 2024-07-19:
> @DieBorr Would you want to look at this? I think this is what I noticed in #11590

Okay, no problem, I Will take a look as soon as I can, probably on this Monday I can take some time for this Issue. 👍

PR #11697 (2024-08-07): Fix for #11687 (Joystick out of screen) + misclick
Fixes #11687. On large screens, the joystick could appear off-screen. Additionally, the reCenter offset has been improved for small screens. Now, there is more offset on the side that is near the center, as there are no widgets there and is more easy to get there with the thumbs, the offset of the side next to the screen boundaries has been decreased.

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2024-08-07 by HTRamsey