mavlink/qgroundcontrol #11693: Multi Vehicle Panel Overhaul

demirciAhmet, 2024-07-20:
Hello, I was working on a QGC custom build for a while (mostly on the UI/UX part), and I would now like to contribute to the development of QGC itself. The current multi-vehicle panel in QGroundControl provides basic functionality for monitoring multiple vehicles. However, I think it lacks certain features and usability improvements that could significantly improve the user experience and operational efficiency. Despite the challenges in making comprehensive changes due to the coupling of widgets and the complexity of QGC, I believe it is reasonable to improve the multi-vehicle panel.
### Issues with the Current Multi Vehicle Panel

- The current panel's design does not align well with QGC's overall Fly View design.
- The widgets on the panel are prone to breaking.
- The usability of the panel is not comprehensive. 
- The multi-vehicle panel selector only appears if more than one vehicle is connected, as mentioned in issue [#8076](https://github.com/mavlink/qgroundcontrol/issues/8076). I think the user should be able to see QGC's capability to control multiple vehicles without needing to connect multiple vehicles.
- Actions initiated through the multi-vehicle panel execute without confirmation, which can lead to accidental commands.

### Suggestions

**UI/UX Improvements**:

- Implementing a new solid top-right panel design that is flexible and easier to enhance.
- Implementing a design that aligns better with the overall Fly View.
- Simplifying instrument widgets for multi-vehicle panel usage.
- Simplified versions of the top bar indicators can be used for multi-vehicle panel usage.

**Functionality Enhancements**:

- Allowing users to change the active vehicle directly from the multi-vehicle panel.
- Enabling customization of displayed information for each vehicle, such as speed and altitude.
- Organizing vehicles based on user-defined criteria.
- Introducing sorting and filtering options.
- Adding additional notification indicators for critical events like low battery or signal loss.

**Usability Enhancements**:

- Ensuring that actions initiated through the panel require confirmation to prevent accidental commands.

[image]
(The mockup design based on Stable_V4.3, and only covers the *solid top-right panel* suggestion above)

What do you think about this concept design? I am willing to implement this for QGC, and would appreciate your thoughs to ensure the successful implementation of these enhancements. Thank you for considering this feature request.

lebronkey, 2024-07-20:
This is a multi machine control station, and I think the multi machine list should be optimized

DonLakeFlyer, 2024-07-22:
Mostly all makes sense to me. Thanks for the help.

Things I worry about:
* "Simplified versions of the top bar indicators can be used for multi-vehicle panel usage." I wonder if this is good or bad. It won't let you see details for the currently active vehicle. That might be something you want, if you want to check on something going wrong with a specific vehicle.
* "I think the user should be able to see QGC's capability to control multiple vehicles without needing to connect multiple vehicles." As long as this doesn't introduce added top-level complexity in the UI for the 99% of people who don't use this.

Suggestions for working on this:
* Take things one small step at time with pulls. This makes code reviews much easier as well as commenting on UI direction before too much work has been put into it.
* Make sure you follow the guidelines with respect to coding style and font size based positioning as opposed to fixed pixel positioning.

closed 2025-01-18 by demirciAhmet