mavlink/qgroundcontrol #11695: The latest QGroundControl and the latest PX4 firmware parameters do not match.

Filipe-Cavalheiro, 2024-07-22:
## Current Behavior
Failed to simulate advanced_plane or any other fix wing vehicle

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. `$ make px4_sitl gz_advanced_plane`
2. open QGC

## System Information
- Operating System: Ubuntu 22.04.4 LTS
- QGC Version: 4.4.0
- QGC build: release
- Flight Controller: Simulation?
- Autopilot (with version): PX4 sitl (if this is not correct please let me know)

## Detailed Description
This bug report is the same as[#11252 ](https://github.com/mavlink/qgroundcontrol/issues/11252) but i can´t see how [#11425](https://github.com/mavlink/qgroundcontrol/pull/11425) fixes it. The bug was reported and fixed in version 4.3.0 yet i still occur the same error

## Log Files and Screenshots
[image]

HTRamsey, 2024-07-22:
That's because that commit wasn't included in 4.4 as far as I can tell. It is still in the daily and will come with the next release

DonLakeFlyer, 2024-07-22:
@julianoes Is new 4.4 still in the works? I could bring this over

julianoes, 2024-07-22:
I was going to publish v4.4.1 soon with all commits on the Stable_V4.4 branch.

julianoes, 2024-07-22:
@HTRamsey which commit is missing?

DonLakeFlyer, 2024-07-22:
This. pull: https://github.com/mavlink/qgroundcontrol/pull/11425

julianoes, 2024-07-26:
@Filipe-Cavalheiro any chance you can test the backport? https://github.com/mavlink/qgroundcontrol/pull/11707

Filipe-Cavalheiro, 2024-07-26:
@julianoes I'm having trouble compiling the code with QT Creator i am following the steps as shown in [https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/index.html](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/index.html) but doing
`git clone -b Stable_V4.4 https://github.com/mavlink/qgroundcontrol.git`  
instead of 
`git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git`

the error is as shown below
[image]

what am i doing wrong? 
PS: Thanks for the help with the rest!

julianoes, 2024-07-26:
I'm sorry, I don't have time to debug your build but you should be able to get pull request artifacts here:

https://github.com/mavlink/qgroundcontrol/actions/runs/10104612249

[image]

Filipe-Cavalheiro, 2024-07-26:
Thank you so so much for everything
happy coding.

closed 2024-07-26 by Filipe-Cavalheiro

HTRamsey, 2024-07-26:
@Filipe-Cavalheiro it's probably best to use qmake for v4.4 instead of cmake.