mavlink/qgroundcontrol #11702: Video lag builds up with Viewlink A10 pro camera

ZephyrinGrilli, 2024-07-25:
## Expected Behavior
stable video feed 

## Current Behavior
after 7-10 minutes, the lag starts to build up exponentially
after around 20 minutes the feed resets and the lag goes away but comes back
closing the app resets the feed and "lag timer"
X axis is time in seconds
y axis is lag in seconds
ignore the yellow line, it was dud test
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. A10 pro running rtsp feed
2. wait around 10 minutes and the lag will be noticable and it builds up to minutes of lag

## System Information
When posting bug reports, include the following information
- Operating System: Android on H16 pro
- QGC Version: 4.2.4 32 bit
-Camera Viewlink A10 Pro FW:S20240103

## Detailed Description
-We've tested the viewlink q10 camera with no issues
-tried the latest stable build 32 and 64 bit but the a10 feed doesn't come up, but the h16 fpv feed works

-tried with no success:
-different camera controller module
-set the QGC max storage from 2048mb to 100mb

-on the camera:
-set the stream type from h264 to h265 but had no video
-fps from 60 to 30
-rtsp output from default to real time priority
-video compression quality from high to medium
-rtsp encoder bitrate from 2045 to 100

I'm not sure what QGC logging settings to use, let me know and i can get whatever is needed

HTRamsey, 2024-07-25:
You can try the daily build which has a newer gstreamer plugin, but it's been better for some and worse for others. GStreamer is a bit difficult to work with so I am adding an alternative receiver based on QtMultimedia that should be a bit more reliable.

ZephyrinGrilli, 2024-07-25:
@HTRamsey 
bit of a github beginner here, how do i get the daily build?

HTRamsey, 2024-07-25:
https://d176tv9ibo4jno.cloudfront.net/builds/master/QGroundControl.apk

closed 2024-11-30 by HTRamsey