mavlink/qgroundcontrol #11712: Container build failure: git error and fuse mount error

bendyna, 2024-07-26:
## Expected Behavior
container build is successful with `deploy/docker/run-docker.sh`

## Current Behavior
multiple errors during container build

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. `./deploy/docker/run-docker.sh`
2. check errors in output, e.g. `ninja: error: loading 'build.ninja': No such file or directory`

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04.4
- QGC build: from master branch

## Detailed Description
1. `./deploy/docker/run-docker.sh >& docker0.log`
2. [docker0.log](https://github.com/user-attachments/files/16395477/docker0.log)
3. added to Git.cmake `execute_process(COMMAND ${GIT_EXECUTABLE} config --global --add safe.directory "${CMAKE_SOURCE_DIR}")`
4. `./deploy/docker/run-docker.sh >& docker1.log`
5. [docker1.log](https://github.com/user-attachments/files/16395531/docker1.log)
6. added ` --security-opt apparmor:unconfined` to docker run command in `run-docker.sh`
7. Now build is successful

Do these fixes make sense? I don't really understand apparmor one. Also wondering why only I see this issue, looks working for others (https://github.com/mavlink/qgroundcontrol/pull/11660).

HTRamsey, 2024-07-27:
That's interesting, I just checked and it built fine for me. But I do see the issues in docker0.log are unrelated to docker. I'll get a fix out probably later today or tomorrow.

HTRamsey, 2024-07-30:
Sorry, doing it now

closed 2024-09-06 by HTRamsey