mavlink/qgroundcontrol #11713: MAVLink Signing Improvements

HTRamsey, 2024-07-27:
Follow up to #11596.
The initial MAVLink Signing implementation supports basic setting of a user provided secret key and sending the setup_signing message to enable/disable signing. Necessary improvements include:

- Cleaning up the QML implementation of setting key & enabling/disabling signing
- Allow unsigned messages options in QML?
- Verify valid secret key input before allowing signing enabled
- Ability to receive setup_signing message from another component?
- Persistent storage of timestamp
- Remove signature and setup_signing message from logs
- Only send setup_signing if connection is wired?

DonLakeFlyer, 2024-07-28:
Isn't the current implementation also missing incoming packet verification against the signing key?

DonLakeFlyer, 2024-07-28:
> Isn't the current implementation also missing incoming packet verification against the signing key? I can't seem to find that code.

Ok, I "think" it's included auto-magically as part of the internal C mavlink lib implementation.

DonLakeFlyer, 2024-07-28:
So I'm still confused by how this is connected into QGC. Problems I see:
* Mavlink signing settings are application not vehicle settings. Hence they can be set when there is no vehicle connected. The current implementation only calls Vehicle::_sendSetupSigning when the signing key value changes. If I setup the key first with no Vehicle and then connect the Vehicle nothing happens. Seems like there needs to be a specific, send the key to the vehicle button of something in order to get the key over. I don't think trying to make this auto-magic at all is a good thing. You also don't want the key going over an un-secure link by accident just be cause you changed a value. To me this should be ab explicit action taken by the user.
* It seems like setup for a channel to support signing (MavlinkSigning::initSigning() is only setup if Vehicle::_sendSetupSigning is called. Doesn't MavlinkSigning::initSigning need to be called at link connection time for all links if signing is enabled?

DonLakeFlyer, 2024-07-28:
Also not quite sure what should happen on a vehicle with multiple links.

HTRamsey, 2024-07-28:
This might be a bit more of a pain but I'm wondering about having the signing settings in each link settings when you are adding a link. Then check for changes when the link settings are changed to queue sending the setup_signing message until the link is connected or something. Maybe it only sends setup_signing at initial connection not sure. Then you can get what type of connection the link is and adjust some settings based on if it is a secure connection (serial/ethernet based) or wifi/bluetooth based.

DonLakeFlyer, 2024-07-29:
I think it's fine to have a single global signing key for all links. Here is what should happen:
* Only if the user clicks a button in the UI should setup signing be done. This will then persist the key on the vehicle and cause the firmware side to send signed packets from here on out. You don't do this every time you connect to a vehicle.
* When a new link is created, if it isn't a USB link then it should call initSigning if there is a global signing key setup. You don't init signing on USB links so you can clear/change a key on the vehicle when the vehicle is already setup with another key.= which you don't have/forgot.

I almost have it all working ...
[image]