mavlink/qgroundcontrol #11718: Usb uvc skydroid not work qgc v 4.4.0

romucifer, 2024-07-29:
Hello
Firmware ardurover v 4.5.4
Qgc v4.4.0 64bits android
 in the usb drop-down menu does not appear to select usb

HTRamsey, 2024-07-29:
This is fixed in daily

romucifer, 2024-07-29:
> This is fixed in daily

Hello i have test in daily v 4.3.0.701 but not work usb uvc
What is daily work?

HTRamsey, 2024-07-29:
Alright I'll check it out, depending on the driver it uses its VID/PID has might have to be manually specified

romucifer, 2024-07-30:
> Alright I'll check it out, depending on the driver it uses its VID/PID might have to be manually specified

hello in windows 11 VID and PID is USB\VID_18EC&PID_5850&MI_00\6&12986e95&0&0000 
please send me info for input to qgroundcontrol in my android thank you very much

closed 2026-10-04 by github-actions[bot]