mavlink/qgroundcontrol #11728: Holybro H-RTK F9P RTK module causing crash on current master Android build

hitagisol, 2024-07-30:
I'm using the SIYI MK32 GCS and building the current master using Android arm64_v8a as the target.
Currently when I connect the F9P RTK base module to the GCS, the RTK icon is showing up but seems the GCS is not really talking with the RTK module, in both survey-in or fixed base position mode (at least in the console log nothing is showing up when enabled the RTKGPS log catalog. Seems the module is detected because it is showing the correct module as u-blox F9P). Then if I detach the module the application crashed.

## System Information
- Operating System: Android 9.0
- QGC Version: master
- QGC build: self build from source
- Flight Controller: issue happens without connecting to the FC

HTRamsey, 2024-07-30:
RTK on mobile was disabled up until recently. I've just recently updated the serial drivers so that RTK modules can actually connect but haven't finished fixing the actual GPS communication portion for android yet. I imagine nothing happens when you connect with v4.4.0 right?

HTRamsey, 2024-07-30:
Could you look at #11677 and see if this is what happen on v4.4.0 for you?

hitagisol, 2024-07-30:
Hi. I just tried on v4.4.0 and yes I'm getting the "internal error" on
LinkManager too. I noticed that in the past the RTK GPS is disabled by that
__mobile__ guard and I saw it has been reenabled in the current master, so
I'm giving it a try.

HTRamsey, 2024-07-30:
If you get the chance I would appreciate your help to try a couple of things and see what happens

hitagisol, 2024-07-30:
> If you get the chance I would appreciate your help to try a couple of things and see what happens

Yes I could give a try. Any how I only have the SIYI MK32 at my hand as the testing device which is running an Android 9.0. If that is find for the test please let me know.

HTRamsey, 2024-07-30:
Should be fine. The first thing, can you change the line 297 in AndroidSerial.cc like this? A timeout of -1 is being sent when the serial library is expecting a 0.
[image]

hitagisol, 2024-07-30:
> AndroidSerial.cc

It's performing the same as before the modification. Just in case we are aligned I'm doing the modification on commit 34516a7d34b0d7ceab62100cc22ec78264d024a3. 
Another thing I'm not sure if causing trouble is in the qserialport_android.cpp, line 138 `bool QSerialPortPrivate::_writeDataOneShot()` is using the writeBuffer, but I'm not able to find where this buffer is appended. I'm not diving deep in the qiodevice yet so probably I'm missing some points.

HTRamsey, 2024-07-30:
Something else you can try is to look at this PR #10250 and try the change in GPSProvider.cc

hitagisol, 2024-07-30:
> Q_OS_ANDROID
After I add the define guard with Q_OS_ANDROID to remove the `if (_serial->waitForBytesWritten(-1))` line in GPSProvider.cc, the behavior changes. Now sometimes the RTK module goes into the survey-in mode (not always). And when it's doing the survey-in, seems the communication stops after random time (around 5 to 10 seconds) and the "Duration" and "Current Accuracy" do not update anymore. 
Then if I detach the RTK module it causing the application to crash.

hitagisol, 2025-05-05:
Hi @HTRamsey,
I saw that there are quite a bunch merges relates to the serial port so I made a fast test on the current master branch 397d781.
The setup up is same as before, using the SIYI MK32 GCS and Holybro H-RTK F9P base station. However the behavior is same as before, which the QGC crash as soon as trying to communicate with the module on the serial port. Just want to ask if there's any update regarding to this issue?
Thank you in advance

HTRamsey, 2025-05-05:
I haven't looked into it in a while, but I'll take a look at it before v5 is released soon

HTRamsey, 2025-05-15:
I can try it out today, but otherwise not a huge deal considering android rtk has never worked before anyways

DonLakeFlyer, 2025-06-11:
We won't hold 5.0 for this.