mavlink/qgroundcontrol #11738: Current master build fails to run on Linux

AndKe, 2024-08-01:
## Current Behavior
Please describe what happens instead of the expected behavior.

After building master with Qt 6.6.3  on Ubuntu 24.04 I get this (on start):
```
(venv-ardupilot) andre@andre-wks:~/prog/qgroundcontrol/build/staging$ ./QGroundControl 
./QGroundControl: symbol lookup error: ./QGroundControl: undefined symbol: qt_resourceFeatureZstd, version Qt_6
```

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. buld master
2. run
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 24.04
- QGC build: master/HEAD

## Detailed Description
on Discord I got an attempt to help:
```
Looks related to this https://codereview.qt-project.org/c/qt/qtbase/+/538064/2
or whatever target you are running it on doesn't have/support zstd
```
but the solution is not clear to me, it does not specify where/how to add QT_NO_AUTORCC_ZSTD

HTRamsey, 2024-08-01:
Did you install libzstd-dev?

AndKe, 2024-08-02:
yes, of course, and I checked in case that was some new dependency.

```
libzstd-dev is already the newest version (1.5.5+dfsg2-2build1).
```

HTRamsey, 2024-08-02:
try creating the appimage

AndKe, 2024-08-08:
creating the appimage? / did you mean it, how?

kschoelpple, 2024-09-03:
What I have to do on my master build on Kubuntu 22.04 i:
Right click on the build file instead of directly running it. Then select "Run in Konsole" 
It will fail initially with some warning message. 
Then I repeat this and it will run fine.
( I believe a background service isn't running at first startup but is ready by the time you start again)

closed 2024-09-10 by AndKe