mavlink/qgroundcontrol #11747: Building master branch on Windows 10 is failing

fraviofii, 2024-08-02:
## Expected Behavior

Build `qgroundcontrol` correctly on Windows 10.

## Current Behavior

After cloning `qgroundcontrol` on my Windows 10 machine, opening the `CMakeLists.txt` file, launching `cmake` is failling, providing the following output:

````

Running C:\Qt\Tools\CMake_64\bin\cmake.exe -S C:/Users/flavi/draganfly/qgc/qgroundcontrol -B C:/Users/flavi/draganfly/qgc/build-qgroundcontrol-Desktop_Qt_6_6_3_MSVC2019_64bit-Debug in C:\Users\flavi\draganfly\qgc\build-qgroundcontrol-Desktop_Qt_6_6_3_MSVC2019_64bit-Debug.
-- QGC_GIT_BRANCH="master"
-- QGC_GIT_HASH="2517cc297"
-- QGC_APP_VERSION_STR="v4.3.0-712-g2517cc297"
-- QGC_APP_VERSION="v4.3.0"
-- QGC_APP_DATE="2023-11-20T18:46:48-08:00"
-- QGC_APP_VERSION_VALID="0"
-- QGC_APP_VERSION="4.3.0" ; QGC_APP_VERSION_MAJOR="4" ; QGC_APP_VERSION_MINOR="3" ; QGC_APP_VERSION_PATCH="0"
-- CMAKE_VERSION="3.24.2" ; CMAKE_PREFIX_PATH="C:/Qt/6.6.3/msvc2019_64" ; CMAKE_HOST_SYSTEM_NAME="Windows" ; CMAKE_HOST_SYSTEM_VERSION="10.0.19045" ; CMAKE_SYSTEM_NAME="Windows" ; CMAKE_SYSTEM_VERSION="10.0.19045" ; CMAKE_CURRENT_SOURCE_DIR="C:/Users/flavi/draganfly/qgc/qgroundcontrol" ; CMAKE_BUILD_TYPE="Debug" ; CMAKE_TOOLCHAIN_FILE=""
-- QT_VERSION="6.6.3" ; QT_MKSPEC="msvc2019_64" ; QT_LIBRARY_HINTS="/6.6.3/msvc2019_64;C:/Qt/6.6.3/msvc2019_64/lib/cmake/Qt6;C:/Qt/6.6.3/msvc2019_64"
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR) 
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR) 
-- 
-- SDL2 was configured with the following options:
-- 
-- Platform: Windows-10.0.19045
-- 64-bit:   TRUE
-- Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/HostX64/x64/cl.exe
-- Revision: SDL-release-2.30.2-0-gf461d91c
-- 
-- Subsystems:
--   Atomic:	ON
--   Audio:	OFF
--   Video:	OFF
--   Render:	OFF
--   Events:	ON
--   Joystick:	ON
--   Haptic:	OFF
--   Hidapi:	ON
--   Power:	OFF
--   Threads:	ON
--   Timers:	OFF
--   File:	OFF
--   Loadso:	ON
--   CPUinfo:	ON
--   Filesystem:	OFF
--   Sensor:	OFF
--   Locale:	ON
--   Misc:	OFF
-- 
-- Options:
--   SDL2_DISABLE_INSTALL        (Wanted: TRUE): OFF
--   SDL2_DISABLE_SDL2MAIN       (Wanted: TRUE): OFF
--   SDL2_DISABLE_UNINSTALL      (Wanted: TRUE): OFF
--   SDL_3DNOW                   (Wanted: FALSE): OFF
--   SDL_ALSA                    (Wanted: OFF): OFF
--   SDL_ALSA_SHARED             (Wanted: OFF): OFF
--   SDL_ALTIVEC                 (Wanted: ON): OFF
--   SDL_ARMNEON                 (Wanted: OFF): OFF
--   SDL_ARMSIMD                 (Wanted: OFF): OFF
--   SDL_ARTS                    (Wanted: OFF): OFF
--   SDL_ARTS_SHARED             (Wanted: OFF): OFF
--   SDL_ASAN                    (Wanted: OFF): OFF
--   SDL_ASSEMBLY                (Wanted: ON): OFF
--   SDL_ASSERTIONS              (Wanted: auto): auto
--   SDL_BACKGROUNDING_SIGNAL    (Wanted: OFF): OFF
--   SDL_CCACHE                  (Wanted: ON): OFF
--   SDL_CLOCK_GETTIME           (Wanted: OFF): OFF
--   SDL_COCOA                   (Wanted: OFF): OFF
--   SDL_DBUS                    (Wanted: OFF): OFF
[... 106 lines cut ...]
  Build step for libevents failed: 1
Call Stack (most recent call first):
  C:/Qt/Tools/CMake_64/share/cmake-3.24/Modules/FetchContent.cmake:1744:EVAL:2 (__FetchContent_directPopulate)
  C:/Qt/Tools/CMake_64/share/cmake-3.24/Modules/FetchContent.cmake:1744 (cmake_language)
  C:/Qt/Tools/CMake_64/share/cmake-3.24/Modules/FetchContent.cmake:1958 (FetchContent_Populate)
  src/MAVLink/LibEvents/CMakeLists.txt:10 (FetchContent_MakeAvailable)

-- Configuring incomplete, errors occurred!
See also "C:/Users/flavi/draganfly/qgc/build-qgroundcontrol-Desktop_Qt_6_6_3_MSVC2019_64bit-Debug/CMakeFiles/CMakeOutput.log".
See also "C:/Users/flavi/draganfly/qgc/build-qgroundcontrol-Desktop_Qt_6_6_3_MSVC2019_64bit-Debug/CMakeFiles/CMakeError.log".
CMake process exited with exit code 1.

Elapsed time: 01:46.
```

## Steps to Reproduce:

1. Clone the repository
2. Open QtCreator 11.0.2 (Community)
3. Open the file `CMakeLists.txt` located at project root directory
4. Run the option `Run Cmake` when pressing the pop-up options for the project

## System Information

More detailed information is available on cmake output presented above

HTRamsey, 2024-08-02:
https://stackoverflow.com/questions/22575662/filename-too-long-in-git-for-windows

fraviofii, 2024-08-02:
Thank you @HTRamsey ,

I executed `git config --global core.longpaths true` on my Windows command prompt than launched again on Qt Creator and it worked.

closed 2024-08-02 by fraviofii