mavlink/qgroundcontrol #11750: Operator id invalid: FRA-RP-XXXXXXXXXXXX

Ostefanini, 2024-08-03:
QGC: 4.4.0rc3

## Expected Behavior
Accept my operator id
## Current Behavior
Says « Operator id invalid »
## Steps to Reproduce:
Enable remoteID
Connect drone to QGC
Set operator id to « FRA-RP-XXXXXXXXXXXX » (where X are digits)
## System Information
Macos  sonoma 14.6
Px4 1.14.3dev
Pixhawk 2.4.8 pro (fmuv3)
## Detailed Description
I’ve obtained my operator id from french platform lphatango

DonLakeFlyer, 2025-01-13:
See if a daily build has the same problem

mahyarmotie, 2025-01-13:
> See if a daily build has the same problem

I have a same problem with dayli build and and android an os vertion too

DonLakeFlyer, 2025-01-13:
Daily build: Go to App Settings / Console / Set Logging. Turn on RemoteIDManagerLog. Go back into remote id settings. Post the log out put here.

mahyarmotie, 2025-01-14:
> What sort of operator ID is this?
> 
> This is how it should look like and work: [#10970 (comment)](https://github.com/mavlink/qgroundcontrol/pull/10970#issue-2082903812)
> 
> Note that you have to add the 3 letter checksum/secret at the end for validation.

thank you for this explanation, it's very helpfull, but i have a problem, with one of our ID we type at the end "-123" and it works but it doesn't work with our another ID , how can we calculate the checksum or 3digit secret?

Ostefanini, 2025-01-14:
> thank you for this explanation, it's very helpfull, but i have a problem, with one of our ID we type at the end "-123" and it works but it doesn't work with our another ID , how can we calculate the [...]

@mahyarmotie where did you find the european operator id format ?

mahyarmotie, 2025-01-14:
FROM ALPHATANGO SITE: 
https://alphatango.aviation-civile.gouv.fr
[image]

Ostefanini, 2025-01-14:
@mahyarmotie thank you !

ianare, 2025-12-26:
Using version 5.0.6 and this does not appear to work:

[image]

Console log:
```
  1059.551 - debug: Arm status error: "bad SYS  dataOP_ID " (RemoteIDManagerLog:unknown:0)
  1060.545 - warning: qrc:/qml/QGroundControl/Controls/AppMessages.qml:47:25: Unable to assign [undefined] to QString (default:unknown:47)
```