mavlink/qgroundcontrol #11758: Corridor scan doesn't properly set camera triggers depending on entry point rotation.

mwrightE38, 2024-08-07:
Sorry for not following the format requested but I wanted to get this reported quickly and have a fix posted for whoever needs it rather than not do it at all.

Corridor scans don't properly set entry/exit coordinate types when the vertices are rotated. This currently causes camera triggers to occur around turns instead of inside the missions every other leg due to the 

[image]

Add lines 335-340 into corridorscancomplexitem.cc for a fix. 

`                    //flip entry and exit coordinates
                    if (reversedVertices.first().coordType == CoordTypeSurveyEntry) {
                        reversedVertices.first().coordType = CoordTypeSurveyExit;
                    } else if (reversedVertices.first().coordType == CoordTypeSurveyExit) {
                        reversedVertices.first().coordType = CoordTypeSurveyEntry;
                    }`

DonLakeFlyer, 2024-08-20:
Was this on Stable or Daily? Daily already has this fix.

mwrightE38, 2024-08-20:
@DonLakeFlyer Sorry about that. I did the fix on an older QGC version so my line numbers are off. It needs added at line 324 in stable as well. The fix that's in stable didn't fully fix the issue only half of it. Probably a more elegant solution than doubling the code in two places but the fix above works. If you can't recreate it or still need more info let me know. 

https://github.com/mavlink/qgroundcontrol/blob/8a9f99e893aa308808b2f21cecabcf0332142fae/src/MissionManager/CorridorScanComplexItem.cc#L324

closed 2026-10-24 by HTRamsey