mavlink/qgroundcontrol #11766: Audio output should be muted while running unit tests

DonLakeFlyer, 2024-08-12:
Looks like this changed with recent AudioOutput changes

DonLakeFlyer, 2024-08-12:
I tried fixing this myself by modify QGCApplication to setMuted. But it didn't work for some strange reason.

HTRamsey, 2024-08-13:
That's because _initForNormalAppBoot doesn't get run during unit tests

DonLakeFlyer, 2024-08-13:
Ah, of course. Forgot about that. That what I get for not debugging it. Got fooled by the unit test check in there. Could you get rid of that as well. Also would be good for this Audio setup code to be closer to the other Audio code. Can it just go into AudioOutput?

HTRamsey, 2024-08-13:
You mean putting the audioMuted connection into AudioOutput? If so then yeah sure, the reason I didn't do that was to not have to link the QGC library to the AudioOutput library, but it still gets linked indirectly through the Utilities library anyways so it doesn't change anything really.

DonLakeFlyer, 2024-08-13:
> You mean putting the audioMuted connection into AudioOutput?

Right. That way all the code is together.

DonLakeFlyer, 2024-08-16:
Now that I know how to fix this I can deal with it myself. It's making me a bit crazy when I run unit tests locally :)

PR #11776 (2024-08-17): Unit test fixes
* Fix #11766
* Fix PlanManager deletion which was causing random crashes in unit tests
closed 2024-08-17 by DonLakeFlyer