mavlink/qgroundcontrol #11799: Allow MAV_CMD_DO_SET_ROI_LOCATION to be sent during a Mission

stryngs, 2024-08-22:
## Expected Behavior
4.4.1 does not allow a user to select a ROI when clicking a location.

## Current Behavior
ROI at location is missing

## Steps to Reproduce:
1. Get airborne
2. Set mode to auto
3. Click on screen

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22
- QGC Version: 4.4.1
- QGC build: https://github.com/mavlink/qgroundcontrol/releases/download/v4.4.1/QGroundControl.AppImage
- Flight Controller: ArduPilot SITL
- Autopilot (with version): ArduPilot Copter 4.5.5

## Detailed Description
A workaround is to use a companion computer and run the following code once airborne and on mission (auto).  The vehicle performs as expected and so I think a code change to qgroundcontrol to allow this would not be unsafe.
```
#!/usr/bin/python3

"""
./sim_vehicle.py -v Rover -f motorboat --console --map --out 127.0.0.1:9898 --out 127.0.0.1:14500
"""

import time
import threading
from pymavlink import mavutil

class Heartbeat():
    """Setup and hold the heartbeat"""
    # __slots__ = ('heartBeat')
    def __init__(self, parent):
        self.heartBeat = threading.Thread(target = parent.heartBeat)
        self.heartBeat.daemon = True
        self.heartBeat.start()

class Parent(object):
    """Parent class responsible for user interactions and backgrounding"""

    def __init__(self):
        # self.args = args
        # self.gList = []
        mport = 9898
        print('[~] Waiting on MAVLink')
        self.m = mavutil.mavlink_connection(f'udpin:%s:%s' % ('127.0.0.1', mport))
        self.m.wait_heartbeat(blocking = True)
        print('[~] MAVLink established')

    def heartBeat(self):
        """Drains the queue for accuracy"""
        while True:
            x=self.m.recv_msg()
            # if x is not None:
            #     print(x)

            time.sleep(.001)

    def setRoi(self, f, lat = -25.363261, long = 49.165230, alt = 20):
        """Set the Region of Interest"""
        msg = f.m.mav.command_long_encode(
            f.m.target_system,     # target_system
            f.m.target_component,  # target_component
            mavutil.mavlink.MAV_CMD_DO_SET_ROI_LOCATION,  # command
            0,                         # confirmation
            0,                         # param1 (ROI mode)
            0,                         # param2
            0,                         # param3
            0,                         # param4
            lat,                   # param5 (latitude)
            long,                   # param6 (longitude)
            alt                    # param7 (altitude)
        )
        f.m.mav.send(msg)

if __name__ == '__main__':

    ## Grab parent obj
    f = Parent()
    Heartbeat(f)
    f.setRoi(f)
```

DonLakeFlyer, 2024-08-22:
The reason behind this is that the thinking is that the Mission has control over the ROI point. And that if the user futzes with the ROI outside of the mission that something strange will happen. Will need to verify with firmware that isn't the case.

stryngs, 2024-08-23:
Look forward to it.  Thank you Don and let me know if you need testing or whatnot.

DonLakeFlyer, 2024-08-24:
I looked at both the PX4 and ArduPilot firmware. Seems like it will be fine to allow this during a mission.

DonLakeFlyer, 2024-10-13:
Pull is up. @stryngs Can you test this?

PR #12009 (2024-10-14): Allow ROI while mission is running
Fixes #11799.
closed 2024-10-14 by DonLakeFlyer