mavlink/qgroundcontrol #11802: The joystick in QGC cannot send buttons information.

1776340526, 2024-08-23:
## Expected Behavior
Solve the problem that the buttons information of the joystick in QGC cannot be transmitted

## Current Behavior
I used mavros (ros2) to connect Raspberry PI 4 and pixhawk4 and connect it to QGC. I connected a game controller in QGC and wanted to read the controller's joystick and button information. Now I can read the controller's x, y, z, r information normally, but the button information is always 0.
In QGC, I can see the button being pressed. In QGC's mavlink check, I found that the buttons message was always 0, and I was confident that the buttons mavlink message was not being sent (Figure 1). However, I have examined the QGC source code, and buttons information can be read all the way up to the packaging and caching operations (Figure 2). In function **Void Vehicle in the Vehicle. Cc file: : sendJoystickDataThreadSafe (float roll, float pitch, float yaw, float thrust, quint16 buttons)** , button information can be sent when I change the order of packaging so that buttons information is placed before **static_cast<int16_t>(newYawCommand)**, But the **newYawCommand** message disappeared (Figure 3). So I think only the first 4 data can be transferred!!

## System Information
When posting bug reports, include the following information
- Operating System:Ubuntu 22.04
- QGC Version: 4.3.0
- QGC build: [e.g.]
- Flight Controller: Pixhawk 4

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots

[image]
(Figure 1)
[image]
(Figure 2)
[image]
(Figure 3)

1776340526, 2024-08-24:
It has bothered me for a long time. Can someone help me?

DieBorr, 2024-09-04:
Hi @1776340526 i have check what you say but I'm not sure what you mean. Here are some screenshots and a short explanation of what did I test and how:

In the first screenshot I show you my set up, with a PS5 controller connected to ubuntu 22.04 running arducopter SITL, in qgc and ardupilot I wrote some code for reading and listening **mavlink_manual_control_t** messages. In qgc I just make a debug of all messages, the screenshots shows the ones with id 69, which is the important one in this topic, on ardupilot part I just made a listening of  **MAVLINK_MSG_ID_MANUAL_CONTROL.** showing up the buttons parameter. The rest screenshots are just for showing you the code I have used for trying to recreate your issue.
On qgc part, at the end of _bool Vehicle::sendMessageOnLinkThreadSafe(LinkInterface* link, mavlink_message_t message)_
```
    qDebug() << "ID:" << message.msgid;
    mavlink_manual_control_t controlMessage;
    mavlink_msg_manual_control_decode(&message, &controlMessage);
    qDebug() << "X:" << controlMessage.x;
    qDebug() << "Y:" << controlMessage.y;
    qDebug() << "Z:" << controlMessage.z;
    qDebug() << "R:" << controlMessage.r;
    qDebug() << "buttons:" << controlMessage.buttons;
```
On ardupilot part, in GCS_Mavlink.cpp (arducopter's one):
```
void GCS_MAVLINK_Copter::handle_manual_control_axes(const mavlink_manual_control_t &packet, const uint32_t tnow)
{
    if (packet.z < 0) { // Copter doesn't do negative thrust
        return;
    }
    hal.console->printf("%d",packet.buttons);
    manual_override(copter.channel_roll, packet.y, 1000, 2000, tnow);
    manual_override(copter.channel_pitch, packet.x, 1000, 2000, tnow, true);
    manual_override(copter.channel_throttle, packet.z, 0, 1000, tnow);
    manual_override(copter.channel_yaw, packet.r, 1000, 2000, tnow);
}
```
As you can see in the next screenshots I can see that the button parametter that is thrown by qgc is captured by SITL
[image]
[image]

Hope that i have understood what did you try to mean / do.

PD: As far as I know mavlink inspector only shows the messages that comes from the vehicle, so probably what u show in a picture does not correspond to the same message, I recommend you to try what i have done in vehicle side, I haven't test it with a companion computer but it's the same thing, as long that you don't overwrite the mavlink message parameters.

1776340526, 2024-09-06:
> 嗨，我检查了你说的话，但我不确定你的意思。以下是一些屏幕截图和我测试了什么以及如何测试的简短说明：
> 
> 在第一张截图中，我向您展示了我的设置，将 PS5 控制器连接到 ubuntu 22.04，运行 arducopter SITL，在 qgc 和 ardupilot 中，我编写了一些代码来读取和收听**mavlink_manual_control_t**消息。在 qgc 中我只是对所有消息进行了调试，截图显示了 id 为 69 的消息，这是这个话题中很重要的，在 ardupilot 部分我只是听了 **MAVLINK_MSG_ID_MANUAL_CONTROL。**显示了 buttons 参数。其余的屏幕截图仅用于向您展示我用于尝试重新创建您的问题的代码。在 qgc 部分，在 _bool Vehicle：：sendMessageOnLinkThreadSafe（LinkInterface* link， mavlink_message_t message）_ 的末尾
> 
> ```
>     qDebug() << "ID:" << message.msgid;
>     mavlink_manual_control_t controlMessage;
>     mavlink_msg_manual_control_decode(&message, &controlMessage);
>     qDebug() << "X:" << controlMessage.x;
>     qDebug() << "Y:" << controlMessage.y;
>     qDebug() << "Z:" << controlMessage.z;
>     qDebug() << "R:" << controlMessage.r;
>     qDebug() << "buttons:" << controlMessage.buttons;
> ```
> 
> 在 ardupilot 部分，在 GCS_Mavlink.cpp 中（arducopter 的 one）：
> 
> ```
> void GCS_MAVLINK_Copter::handle_manual_control_axes(const mavlink_manual_control_t &packet, const uint32_t tnow)
> {
>     if (packet.z < 0) { // Copter doesn't do negative thrust
>         return;
>     }
>     hal.console->printf("%d",packet.buttons);
>     manual_override(copter.channel_roll, packet.y, 1000, 2000, tnow);
>     manual_override(copter.channel_pitch, packet.x, 1000, 2000, tnow, true);
>     manual_override(copter.channel_throttle, packet.z, 0, 1000, tnow);
>     manual_override(copter.channel_yaw, packet.r, 1000, 2000, tnow);
> }
> ```
> 
> 正如您在下面的屏幕截图中看到的那样，我可以看到 qgc 抛出的按钮参数被 SITL 捕获 ![image](https://private-user-images.githubusercontent.com/47715559/364442035-cd49fedf-24d1-4334-b1cd-9437ab371c72.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1ODkxMjIsIm5iZiI6MTcyNTU4ODgyMiwicGF0aCI6Ii80NzcxNTU1OS8zNjQ0NDIwMzUtY2Q0OWZlZGYtMjRkMS00MzM0LWIxY2QtOTQzN2FiMzcxYzcyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA2VDAyMTM0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFlNTRhNDg5ZjEzOWE5NGVlZDNkMmQ5NjY2Y2NhNTc5NTBjZTdmZDNjNDNkM2IxMTI1MDcyOGYxOWZlOGYwZjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8ZMGY53re1uN90bG026vEEvgDKKVcWkRyXv-BL7hOgQ) ![image](https://private-user-images.githubusercontent.com/47715559/364450965-910f53e0-3b3e-40cd-9cd5-2d8b679cf9bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1ODkxMjIsIm5iZiI6MTcyNTU4ODgyMiwicGF0aCI6Ii80NzcxNTU1OS8zNjQ0NTA5NjUtOTEwZjUzZTAtM2IzZS00MGNkLTljZDUtMmQ4YjY3OWNmOWJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA2VDAyMTM0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlYzdjZWE5NzExOWMyZjE4YjMzNDkzMjZlZjJkNTVkYjZiZGRmMTJmMjU2ZDM5OTYxYWNjMWYyYTc2Nzc4NzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6IYDq6rlJyV6VDS4V3Qq6nfgcFGHqD-v12UIh7EfbkU)
> 
> 希望我已经理解了你试图表达/做什么。
> 
> PD：据我所知，mavlink 检查器只显示来自车辆的消息，所以可能在图片中显示的与相同的消息不对应，我建议你试试我在车辆方面所做的，我没有用机载电脑测试过，但这是一回事， 只要你不覆盖 mavlink 消息参数。

Thank you. I've already solved the problem. It turns out that the problem is with mavros, whose handler filters the mavlink message sent by qgc and receives the message returned by fcu, so the button message is removed.