mavlink/qgroundcontrol #11803: Septentrio base station static position

SeptentrioGNSS, 2024-08-23:
Please only ask questions related to source code.
- For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources
- If your question still isn't answered, please check the forums: https://discuss.px4.io/c/qgroundcontrol/15.
- If it is about Qt or any QGC dependencies, please refer to them instead.

While trying the latest FW (4.4.1) of QGC, I noticed that the parameters of the RTK GPS (specific base position) Lat, Lon, Elev do not get send to the receiver. Could it be that these do not get correctly tranfered to the GPS driver where it is sending the command from? I get part of the command through to my Septentrio receiver but not the final three values.

Talking about the code in the GPS driver where the values should come from BaseSettingsType:::fixed_position and the snprintf(msg, sizeof(msg), SBF_CONFIG_RTCM_STATIC_COORDINATES)
[image]

SeptentrioGNSS, 2024-08-23:
@julianoes could you shed some light on the fact that in the time this code was implemented and now there have been updates to where thes fixed base station coordinate values have been stored?

julianoes, 2024-08-25:
I guess I don't understand RTK. Isn't the purpose of RTK to send RTCM corrections from the base to the drone's GPS? Or why would we also need lat/lon, etc.?

SeptentrioGNSS, 2024-08-26:
That is indeed the purpose of RTK, but the backstory behind this is a bit more complex. The quality of RTCM messages (higher quality means higher accuracy) can only be derived from a base station that, to a very high precision, knows it's own location. In QGC there are two options to get to this high accuracy position. First one is the Survey-in option, here the GCS sends the necessary messages/commands to the base station in order to perform a position estimation with both a duration and accuracy requirement as parameters. On the other hand, you can enter LAT, LON and height as parameters if you know them yourself (can be done through a number of different methods, example of list [here](https://customersupport.septentrio.com/s/article/How-to-set-up-a-base-station-Chapter-4-Determining-the-reference-position)). This, if done correctly, is always the more accurate way to go. 

But the issue I am currently having is that on the Septentrio side of this fixed position base implementation, the values for latitude, longitude and height do not get passed to the Septentrio command (in code above the SBF_CONFIG_RTCM_STATIC_COORDINATES) that is sent to the receiver that is a base station. With this, the stream of RTCM into the GCS does not start and the receiver is just doing nothing. 

What I ask is someone to look into this who has more background on how the GPS driver has gotten to this stage and might find the solution why the _base_settings.settings.fixed.position.latitude/longitude/altitude might be empty or not included in this specific line of the code.

julianoes, 2024-08-26:
I see. I was assuming @flyingthingsintothings had been fixing up what I had done as required but it doesn't sound like it.

I can offer to have a closer look into this but I'm afraid, I'm completely booked out by other work at the moment. It would have to be paid by Septentrio and/or Holybro (FYI @vincentpoont2) who sponsored the previous fixes.

SeptentrioGNSS, 2024-08-27:
He did indeed but this is an issue that just came to the surface. There have been done improvements on this indeed but seems like this functionality was never fully fixed or has been changed in recent times causing it to break. What I ask is just a glance at the path of travel of these parameters and why they might not end up where they should. I do not feel like this should come with a paid endeavour.

closed 2026-10-05 by github-actions[bot]